OpenTabletDriver/OpenTabletDriver #1717: Wacom PTZ-x30/31W FeatureInitReport does not work properly

Kuuuube, 2021-12-11:
## Description

The current Wacom PTZ-x30/31W does not actually init the tablet in any working way. Inits are accepted but they seem to go nowhere. 

`AgI=` init alone does init the tablet properly into 100hz mode. `AgI=` as the last init sent regardless of inits before it does init the tablet properly into 100hz mode. Any combination of inits that does not put `AgI=` as the last init or as the only init, does not init the tablet at all.

Sending `AgI=` alone then `BAA=` alone (detecting tablet between two different configs without replugging) OR plugging in the tablet then sending `AgI=` and `BAA=` together (detecting tablet with a two configs the first with no init the second with both) inits the tablet properly into 200hz mode. 

`BAA=` does nothing on its own and must be paired with `AgI=`.

`CAIDAAA=` appears to do nothing regardless of what it is paired with.

The current config seems to be sending inits too quickly after plugging in the tablet and they get nullified somehow.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| OpenTabletDriver Version | 4b9e63f
| Tablet                   | Wacom PTZ-x30/31W

gonX, 2021-12-25:
Would a config attribute enabling delayed send of multiple inits solve this issue? And does the tablet respond with anything after the first init that we can wait for?

gonX, 2021-12-26:
Talked with Kuuube about it, sounds like if we have to meet this for 0.6.x, simply delaying additional reports by a second should be fine - the side effect is really just a slight delay in activating higher poll rates.

Kuuuube, 2021-12-26:
Not sure if this was conveyed properly in the initial post but if this is not fixed people using Wacom PTZ-x30/31W tablets will not have a working tablet. As a workaround they can detect tablet a second time every time they start OTD but that is very annoying. 

Alternatively, configs could be stripped of `BAA=` and `CAIDAAA=` inits and let the tablet go into 100hz mode instead of trying for 200hz. This will allow for the tablets to work without any extra action from the user but only 100hz... that option isn't great but **something** needs to be done about this before 0.6.0 releases.

Kuuuube, 2022-01-06:
Putting this back on the 0.6.0 milestone. 

As I mentioned in my previous comment something needs to be done about this before 0.6.0. Even if that means we revert back to only using `AgI=` inits for the Wacom PTZ-x30/31W. If this done then this issue can be put back on the backlog.

However, a proper fix involving a delay between inits or waiting for the tablet to send data back after the first init would be best and this issue shouldn't be closed until then.

InfinityGhost, 2022-01-07:
PR to revert the init is needed then. Don't link it as closing, but reference this issue in some way.

Kert, 2022-01-14:
This doesn't happen on my PTZ-1231W. Multiple inits go through without delay fine and tablet is set to 200hz mode correctly