OpenTabletDriver/OpenTabletDriver #1718: Race condition when launching OTD.Daemon and OTD.UX at the same time

X9VoiD, 2021-12-11:
## Description

Launch OTD.UX and OTD.Daemon at the same time. Doing so will produce this stack trace:
```
[NullReferenceException:Error]	Object reference not set to an instance of an object.
   at OpenTabletDriver.Devices.DeviceReader`1.Main() in /home/oscar/Work/OpenTabletDriver/OpenTabletDriver/Devices/DeviceReader.cs:line 106
Unhandled exception. System.InvalidOperationException: Collection was modified; enumeration operation may not execute.
   at System.Collections.Generic.List`1.Enumerator.MoveNext()
   at System.Linq.Enumerable.SelectListIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Collections.ObjectModel.ObservableCollection`1.CreateCopy(IEnumerable`1 collection, String paramName)
   at System.Collections.ObjectModel.ObservableCollection`1..ctor(IEnumerable`1 collection)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 profiles) in /home/oscar/Work/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/ProfileCollection.cs:line 17
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 tablets) in /home/oscar/Work/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/ProfileCollection.cs:line 22
   at OpenTabletDriver.Desktop.Settings.GetDefaultProfiles() in /home/oscar/Work/OpenTabletDriver/OpenTabletDriver.Desktop/Settings.cs:line 59
   at OpenTabletDriver.Desktop.Settings.GetDefaults() in /home/oscar/Work/OpenTabletDriver/OpenTabletDriver.Desktop/Settings.cs:line 49
   at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings() in /home/oscar/Work/OpenTabletDriver/OpenTabletDriver.Daemon/DriverDaemon.cs:line 189
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings() in /home/oscar/Work/OpenTabletDriver/OpenTabletDriver.Daemon/DriverDaemon.cs:line 226
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__127_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
Aborted (core dumped)
```

gonX, 2021-12-11:
Can you give better repro steps and check which commit you're on? Doing `./OpenTabletDriver.Daemon & ./OpenTabletDriver.UX.Gtk` does not cause any issues for me on `4b9e63f58ba5c1a`

X9VoiD, 2021-12-11:
Settings issue, there's no thread safety in migration, the Daemon is regenerating new settings while UX reads it.

gonX, 2023-03-28:
Fixed by #2552

PR #2476 (2022-11-14): Resolve race condition bugs of Driver.InputDevices
Use immutable array and atomically update InputDevices.

Fixes ppy/osu#21078
Fixes #1718
Fixes #2228