OpenTabletDriver/OpenTabletDriver #1719: NRE on any form of input

X9VoiD, 2021-12-11:
## Description

Reproduction:
- Do anything with the tablet.

```
[NullReferenceException:Error]	Object reference not set to an instance of an object.
   at OpenTabletDriver.Devices.DeviceReader`1.Main() in (schtuff)/OpenTabletDriver/Devices/DeviceReader.cs:line 106
```

gonX, 2021-12-11:
Can't reproduce. Commit id and OS?

andigandi, 2021-12-11:
Is this related to #1703 ? Are you using dotnet 6?

X9VoiD, 2021-12-11:
> Can't reproduce on [4b9e63f](https://github.com/OpenTabletDriver/OpenTabletDriver/commit/4b9e63f58ba5c1a271daa995acdf51eb2fb623de). What your commit id and OS?

4b9e63f and Arch Linux

> Is this related to #1703 ? Are you using dotnet 6?

Not related, this is from a clean build with zero settings. And OTD.Daemon/UX only builds with .NET 6.

X9VoiD, 2021-12-12:
It's an issue with the build script not clearing out the ./bin folder.

closed 2021-12-12 by X9VoiD