OpenTabletDriver/OpenTabletDriver #1722: Add support for HUION Kamvas 20

jinkhya, 2021-12-12:
Hello, and thanks in advance :)

# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | HUION
| Product Name    | KAMVAS 20 ( GS1901) 
| Width           | 435mm
| Height          | 239mm
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information
[OpenTabletDriver.Console.deps.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7699325/OpenTabletDriver.Console.deps.txt)

Mrcubix, 2021-12-12:
Hi, can you go to `Tablets > Device string reader` and enter the following info:

- 9580
- 110
- 200

then click on send request and send us back the output string

gonX, 2022-01-18:
This is now supported in 0.6.x

PR #1731 (2021-12-30): Added Kamvas 20 Configuration
# Verification:

https://discord.com/channels/615607687467761684/645408221322018837/919662139180974181

# Changes:

- Fixes #1722
closed 2022-01-18 by gonX