OpenTabletDriver/OpenTabletDriver #1732: Tablet Debugger throws exceptions galore

gonX, 2021-12-15:
## Description

Tablet debugger does not work _at all_. The report count will increase by 1 every time the tablet is brought in range, but is actually not doing anything except giving me a lot of exception popups.

```
[InvalidOperationException:Error]    Sequence contains no matching element
   at System.Linq.ThrowHelper.ThrowNoMatchException()
   at OpenTabletDriver.Desktop.RPC.DebugReportData.ToObject() in /home/gonx/git/OpenTabletDriver/OpenTabletDriver.Desktop/RPC/DebugReportData.cs:line 28
   at lambda_method179(Closure , DebugReportData )
   at Eto.Forms.DelegateBinding`1.get_DataValue()
   at Eto.Forms.DirectBinding`1.<>c__DisplayClass14_0`1.<Child>g__valueChanged|0(Object sender, EventArgs e)
   at Eto.Forms.DirectBinding`1.<>c__DisplayClass14_0`1.<Child>g__valueChanged|0(Object sender, EventArgs e)
   at OpenTabletDriver.UX.Windows.Tablet.TabletDebugger.OnReportDataChanged() in /home/gonx/git/OpenTabletDriver/OpenTabletDriver.UX/Windows/Tablet/TabletDebugger.cs:line 214
   at OpenTabletDriver.UX.Windows.Tablet.TabletDebugger.<>c__DisplayClass41_0.<HandleReport>b__0() in /home/gonx/git/OpenTabletDriver/OpenTabletDriver.UX/Windows/Tablet/TabletDebugger.cs:line 274
   at Gtk.Application.InvokeCB.Invoke()
   at GLib.Timeout.TimeoutProxy.Handler()
```

I'm guessing it happened during the `configs` merge, but don't quote me on that.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Arch Linux
| OpenTabletDriver Version | 2850f10
| Tablet                   | Probably all, but confirmed on GD-0608 and XP-Pen G640

gonX, 2021-12-15:
```git bisect bad
dd98f6f7183e512414e7f4b02c762d6a4e7b9367 is the first bad commit
commit dd98f6f7183e512414e7f4b02c762d6a4e7b9367
Author: X9VoiD <oscar.silvestrexx@gmail.com>
Date:   Thu Nov 4 13:11:13 2021 +0800

    Move vendor parsers to OTD.Configurations
```

InfinityGhost, 2021-12-15:
DebugReportData relies on reflection, likely the same issue source for #1730. Likely early loading the assembly (before PluginManager initializes) would resolve this.

PR #1739 (2021-12-17): PluginManager: Force Configurations and Native assemblies to be loaded
Patch fix for loading required assembly for plugins

- Closes #1730
- Closes #1732
closed 2021-12-17 by InfinityGhost