OpenTabletDriver/OpenTabletDriver #1741: Add support for Acepen AP 604

Henry-YSLin, 2021-12-16:
Additional information to #451
Thanks.

# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Acepen |
| Product Name    | AP 604 |
| Width           | 6" |
| Height          | 4" |
| Pressure Levels | 8192 |
| LPI             | 5080 |

## Diagnostic Information

[acepen.ap604.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7724105/acepen.ap604.json.txt)

Mrcubix, 2021-12-16:
It seem like this tablet will only require WinUSB to be removed.
To proceed, do the following:
- Right click on the start menu > Device manager
- View > Device by container
- Look for your tablet there and open it's container (arrow on the left)
- Screenshot the content of that container so we can proceed

Henry-YSLin, 2021-12-17:
[image]

Henry-YSLin, 2021-12-17:
So... I found a new version of the official driver on acepen's website, uninstalled the old driver for that and found out that the new driver is completely useless (does not detect the device at all). Now I can't go back to the old driver because the installer is lost...

Here's how it looks like now, with no drivers installed:
[image]

Can I proceed without the old, working driver?

Mrcubix, 2021-12-17:
export another diag

Henry-YSLin, 2021-12-17:
[diag2.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735406/diag2.json.txt)

Mrcubix, 2021-12-17:
the device does no seem to be plugged in or hasn't been replugged since uninstall

Henry-YSLin, 2021-12-17:
Oops, sorry. Cable came loose
[diag3.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735443/diag3.json.txt)

Mrcubix, 2021-12-17:
`Tablet > device string reader` then enter following:

- 21827
- 97
- leave it empty

then click on dump all and send the generated file here

Henry-YSLin, 2021-12-17:
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735497/dump.txt)

Mrcubix, 2021-12-17:
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735511/AP.604.zip)
try placing this config in the `Configurations` folder then in OTD `Tablet > Detect Tablet`, it should be detected, check if it move, lemme know if it doesn't

Henry-YSLin, 2021-12-17:
[image]

It's detecting but the parsing seems to be wrong

Mrcubix, 2021-12-17:
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735564/AP.604.zip)
try this config

Henry-YSLin, 2021-12-17:
[image]
It's detecting pen up events pretty well but the tablet area and aux buttons are wrong.

Mrcubix, 2021-12-17:
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735588/AP.604.zip)
try this config, re-detect once done

Henry-YSLin, 2021-12-17:
> AP 604.zip
> try this config, re-detect once done

It's treating pen move as aux buttons and pen up as pen move.

> If it doesn't, we'll have to do a USB Capture using the manufacturer driver

Not sure if it's the same, but I captured the tablet input with a USB sniffer before.

Hold on

Mrcubix, 2021-12-17:
- Close OTD,

- Possibly Re-Install the manufacturer drivers,

- Start the manufacturer driver,

- https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool,

- When done, head to `C:\Program Files\USBPcap` in the explorer (you can probably copy paste that path in the path bar),

- Once you're there, start `USBPcapCMD.exe`

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to `C:\Program  Files\USBPcap`,

- Start USBPcapCMD.exe,

- screenshot the console.

Henry-YSLin, 2021-12-17:
[image]

Mrcubix, 2021-12-17:
- type 1 then enter,
- name the file. Press enter and do the following steps:

- Move for 5s
- Unplug the tablet from your computer (remove the usb cable from the port),
- Wait 2 seconds,
- Plug in your tablet again (connect your tablet to your pc ),
- Wait 2 seconds,
- Move for 5s
- Close the console (red cross),
- A new file has been generated with the name you gave in the same folder,
- Upload that file here

Henry-YSLin, 2021-12-17:
[data.pcap.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735681/data.pcap.txt)

Mrcubix, 2021-12-17:
Then it will be impossible to capture init without them, therefore potentially making it impossible to add compatibility for this tablet

Henry-YSLin, 2021-12-17:
I'll see if I can find the old driver then. The new one does nothing.

Mrcubix, 2021-12-17:
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735725/AP.604.zip)
try this config tho i doubt it will take it

Henry-YSLin, 2021-12-17:
> AP 604.zip
> try this config tho i doubt it will take it

tablet detected but no data

Henry-YSLin, 2021-12-17:
[data.pcap.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735763/data.pcap.txt)
Found the driver. This one's done with the driver installed.

Mrcubix, 2021-12-17:
can you screenshot the device in device manager again?

Henry-YSLin, 2021-12-17:
[image]

Mrcubix, 2021-12-17:
right click on the first `DigitizerPanelUpin Tablet Device` > update > second option > let me pick from a list > screenshot

Henry-YSLin, 2021-12-17:
[image]

Mrcubix, 2021-12-17:
install usb input device on both

Henry-YSLin, 2021-12-17:
[diag4.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735836/diag4.json.txt)

Mrcubix, 2021-12-17:
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735848/AP.604.zip)
try this config

Henry-YSLin, 2021-12-17:
[image]
tablet detected but debugger is empty

Mrcubix, 2021-12-17:
Alright, i'm making a new parser and i'm going to have to ask you:

How many programmable express keys are on this piece of electronic waste

Henry-YSLin, 2021-12-17:
4 on the tablet, 2 on the pen that only works when the pen is hovering on the tablet

> this piece of electronic waste

Can't agree more

Mrcubix, 2021-12-17:
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7735912/AP.604.zip)
try this config, i'll need to know which bytes change when pressing express keys and how they change

Henry-YSLin, 2021-12-17:
Nothing happens in OTD when I press express keys (probably because they are 8-byte instead of 10?)

This is when I sniff the traffic with busdog:
4 keys are pressed in sequence
[image]

Mrcubix, 2021-12-17:
try changing the init string in the config to `"200"` instead of `"100"` and see if it make any difference

Henry-YSLin, 2021-12-17:
makes no difference

Mrcubix, 2021-12-17:
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7736126/AP.604.zip)
try this, it contain veikk's output init which was also used by the AcePen 1060

Henry-YSLin, 2021-12-18:
Still the same

Mrcubix, 2021-12-18:
- Close OTD completely
- Get https://github.com/OpenTabletDriver/OpenTabletDriver/suites/4696296263/artifacts/128305799
- Extract it to it's own folder
- Create a `userdata` folder at the same location as the exes
- Create a `Configurations` folder at the same location at the exes
- Extract the content of this zip file inside
[AP 604.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7738461/AP.604.zip)
- Start the UX
- Test your tablet, see if it's any different

Henry-YSLin, 2021-12-18:
OTD crashed when connecting to daemon
[daemon.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7738494/daemon.log)