OpenTabletDriver/OpenTabletDriver #1744: Change Max Pressure for XP-Pen Deco Mini 4.

Ivanoks, 2021-12-16:
## Description

I'm stupid, I already forgot how I wrote the value 2048 then, but the official website says 8192. So... Change please. :)
[image]

## System Information:

I don't think this is necessary.

gonX, 2021-12-16:
Can you test manually whether it reports 8191 or 8192 under full pressure? You can use the Tablet Debugger (OTD, Help -> Tablet Debugger) to test this

Ivanoks, 2021-12-16:
> Can you test manually whether it reports 8191 or 8192 under full pressure? You can use the Tablet Debugger (OTD, Help -> Tablet Debugger) to test this

Oh.. In Tablet Debugger there will be no restriction due to config? And at the moment, my friend who has this tablet is at school right now. Xd

Mrcubix, 2021-12-16:
If the tablet report 2047 pressure then 2047 it is, just gotra check in the debugger as manufacturers often lie about specs

Ivanoks, 2021-12-16:
> If the tablet report 2047 pressure then 2047 it is. Just gotta check in the debugger as manufacturers often lie about specs

Okay, got it. Then I'll wait for my friend.

Ivanoks, 2021-12-16:
> If the tablet report 2047 pressure then 2047 it is. Just gotta check in the debugger as manufacturers often lie about specs

[image]

Okay, I ended up screwing up. That's 8191.

jamesbt365, 2021-12-28:
fixed in #1746

PR #1746 (2021-12-26): Correct the max pressure for the deco 4 mini.
closes #1744
closed 2021-12-28 by jamesbt365