OpenTabletDriver/OpenTabletDriver #1755: Package as AppImage

mochaaP, 2021-12-17:
## Description

This helps solve the dependency issues that were usually encountered on Debian unstable by packaging a .AppImage file.
seealso: <https://github.com/AppImage/awesome-appimage#deployment-tools-for-net-core-mono-applications>

Sublimelime, 2021-12-17:
Re-enabling self-contained publishing would likely fix this issue. It was initially disabled due to concerns about executable size, but there's been murmurs of re-enabling it for 0.6.0. Relevant pr is #1442 .

gonX, 2021-12-18:
> Re-enabling self-contained publishing would likely fix this issue.

I'd prefer if that PR did not close this issue. We should evaluate the value of AppImage for OTD regardless.

gonX, 2022-01-18:
So the issue with packaging OpenTabletDriver as an AppImage is that udev rules are inevitably going to cause pain for us, as the user isn't going to be having direct access to the tablet endpoints we want.

If any AppImage experts can chime in how we can grab hidraw and uinput endpoints without huge fuzz this would be pretty cool.