OpenTabletDriver/OpenTabletDriver #1762: Plugins do not have access to OutputMode

Kuuuube, 2021-12-18:
## Description

Previously (0.5.x) plugins had access to OutputMode though Info. This allowed plugins to get the current output mode (relative or absolute), the current tablet area and the current monitor area settings. 

The previously added TabletReference (#1617) has made it possible to get the tablet's specifications but not the active settings.

Example in Circular Area: https://github.com/Kuuuube/Circular_Area/blob/main/Circular_Area/CircularBase.cs#L12-L31

The following plugins on the current plugin repo cannot be updated:
Circular Area
Tablet Area Randomizer
Gess1t's Area Randomizer

## System Information:

| Name             | Value |
| ---------------- | ----- |
| OpenTabletDriver Version | edcee3d2db1974779f33565b476ea6044caf9c02

gonX, 2025-01-25:
Master abandoned - Avalonia doesn't have plugins yet but this will be fixed when #3706 is implemented

closed 2025-01-25 by gonX

gonX, 2025-10-02:
Reopening as the issue is likely still relevant to Avalonia.

gonX, 2026-09-23:
Moving this to 0.6.x since it seems like the current way is kinda hacky.