OpenTabletDriver/OpenTabletDriver #1764: Error: RemoteMethodNotFound

hXtreme, 2021-12-19:
## Description
I recently tried to switch from xwacom drivers to OpenTabletDriver.
After removing xwacom driver and installing OpenTabletDriver and adding systemd units my tablet is no longer recognized and there's no response when I move the stylus or click the buttons. Furthermore when I open the OpenTabletDriver GUI I receive the following error message:

[image]

The tablet is still recognized to the point where it shows up on `lsusb`

```
$ lsusb
...
Bus 003 Device 007: ID 256c:006d HUION Huion Tablet_HS64
...
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |  Arch Linux (5.15.10-arch1-1) |
| OpenTabletDriver Version | v0.5.3.1-r971.gedcee3d2 |
| Tablet                   | HUION Huion Tablet_HS64 |

Mrcubix, 2021-12-19:
i would advice against using `opentabletdriver-git` (master) right now as it has many issue, you should be able to get OTD from AUR with `opentabletdriver`

hXtreme, 2021-12-19:
Switching to stable seems to have fixed it.

X9VoiD, 2021-12-19:
This error happens when you're running an old daemon and a new ux.

closed 2021-12-19 by X9VoiD