OpenTabletDriver/OpenTabletDriver #1766: HawkuFilters Smoothing

colorgrey, 2021-12-19:
## Description

Sometimes when I turn on smoothing from the "HawkuFilters" plugin and set it to anything other than 0ms my cursor sometimes freezes. I don't know if it is the driver or not because I replug the tablet and restart the driver but it still freezes.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | macOS Mojave 10.14.6
| OpenTabletDriver Version | v0.5.3.3
| Tablet                   | CTL-472

jamesbt365, 2022-02-13:
Is this caused from your cpu jumping high?

jamesbt365, 2022-07-28:
If so its likely going to be resolved by #2342

InfinityGhost, 2022-07-29:
Definitely related to #2342

gonX, 2022-11-07:
Indirect duplicate of #473

PR #2342 (2022-12-26): native timer on macos
- Closes #473 
- Closes #1766
closed 2022-12-26 by X9VoiD