OpenTabletDriver/OpenTabletDriver #1767: Setting Linux Artist Mode should set or notify of Pen Passthrough

gonX, 2021-12-19:
## Description
Right now, switching to Linux Artist mode and not setting pen passthrough for pen tip (and other relevant stylus keys) will cause odd behavior.

It's not obvious for first time users that they're required to set this manually.

Setting this to medium priority because I foresee a few users never really trying OTD because "artist mode didn't work"

zer0exia, 2021-12-23:
This might need to be pinned (or mentioned somewhere). Veikk S640, enabling artist mode straight up made the pen tip unusable without setting the pen passthrough. 

Uninstalled & reinstalled OTB after a while (caused by arch dotnet issue). Would not be obvious for me without reading this post, because it was working fine before I uninstalled.

InfinityGhost, 2021-12-23:
Maybe at least documented for the v0.6 release, ideally on https://opentabletdriver.net

jamesbt365, 2021-12-31:
This should more or less notify that the tip has to be unbound, if that has to be done as if i remember correctly pen pass through was removed and instead tip has to be bound, probably wrong though.

X9VoiD, 2022-12-31:
There's no pen passthrough anymore.

closed 2022-12-31 by X9VoiD