OpenTabletDriver/OpenTabletDriver #1768: Support Request: Before I install it.....

amanda-santini, 2021-12-19:
# Support Request

Hi there!

I'm using a Huion H320M, currently with the DIGImend drivers. However, I can't seem to map the tablet buttons for "keyboard + mouse" action, it's either just keyboard or mouse. For instance, I work professionally with Maya and Substance Painter. On these, camera controls are achieved with Alt + (mouse left, mouse middle, mouse right). On Windows I used to bind the last bottom button as "Alt + Left Mouse", the next button on top as "Alt + Right Mouse", and the pen button 2 as "Alt + Middle Mouse".
I'm not able to do so with the current drivers and manual .conf configuration.

Will I be able to map the buttons this way using your driver?

-------------------------------------------------------------------------------------------------------

## What I've already tried

I can partially achieve what I want. Only problem is that the command isn't permanent until I release the button, it's executed and de-executed instantly.
Command:
xsetwacom set "HUION Huion Tablet_H320M Pad pad" button 1 "key +Alt +button 1"

However, note this:
xsetwacom get "HUION Huion Tablet_H320M Pad pad" button 1
key +Alt_L button +3 -3 

That "-3" is what's probably deactivating the command instantly.

Thanks in advance.

jamesbt365, 2021-12-19:
Multi key bindings are going to be implemented soon but are not availible at the moment

gonX, 2021-12-19:
Yeah, we have multi-key keyboard bindings but not mouse+keyboard bindings yet. We'll track this.

jamesbt365, 2022-06-18:
Closing due to having an issue already tracking this.

closed 2022-06-18 by jamesbt365