OpenTabletDriver/OpenTabletDriver #1773: Установил драйвера, а они не работают

GeromIy, 2021-12-20:
## Description

Драйвера видят планшет, но сам планшет, видимо, не видит драйвер. При изменении любых настроек никакой реакции. Настраиваю область, а планшет работает в заводском режиме (полная область). Как это пофиксить?
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 5.3.3
| Tablet                   | Wacom CTL 471

Mrcubix, 2021-12-20:
деинсталляция wacom

если не установлен
установить, затем удалить

GeromIy, 2021-12-20:
> деинсталляция wacom
> 
> если не установлен установить, затем удалить

Удалил заранее, перед установкой. Комп ребутал

Mrcubix, 2021-12-20:
экспорт диагностического файла

GeromIy, 2021-12-20:
> экспорт диагностического файла

?

Mrcubix, 2021-12-20:
`Help > Export diagnostics`

GeromIy, 2021-12-20:
> `Help > Export diagnostics`

А куда это писать?

GeromIy, 2021-12-20:
> `Help > Export diagnostics`
> 
> [image]

Мне текст из файла скинуть?

Mrcubix, 2021-12-20:
везде

добавить .txt

загрузите файл сюда

GeromIy, 2021-12-20:
[11111.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7742532/11111.txt)

Mrcubix, 2021-12-20:
[image]

как вы удалили wacom

GeromIy, 2021-12-20:
> [image]
> 
> как вы удалили wacom

Через программы и компоненты. Потом открылся сам драйвер, спросил "Вы хотите удалить драйвера wacom?", я нажал "да", удалил и ребутнул пк

Ivanoks, 2021-12-20:
Привет, давай я тебе попытаюсь помочь, скинь Дискорд, я через время тебе напишу. (Я поделаю банальную херню, мб поможет.)

GeromIy, 2021-12-20:
> Привет, давай я тебе попытаюсь помочь, скинь Дискорд, я через время тебе напишу. (Я поделаю банальную херню, мб поможет.)
Geromi#0539

jamesbt365, 2022-01-14:
Going to assume this was resolved due to the last comment being 26 days ago and mentions of going to discord.

closed 2022-01-14 by jamesbt365