OpenTabletDriver/OpenTabletDriver #1774: Add support for HUION P608N

jxu, 2021-12-20:
# Tablet Information
| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Huion
| Product Name    | P608N
| Width           | 8"
| Height          | 6"
| Pressure Levels | 2048
| LPI             | 5080

## Diagnostic Information

```
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_01&col01#8&276916e5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 9580,
      "ProductID": 110,
      "CanOpen": false,
      "ReleaseNumberBcd": 0,
      "ReleaseNumber": "0.0",
      "ProductVersion": 0,
      "Manufacturer": "HUION",
      "ProductName": "P608N",
      "SerialNumber": "",
      "MaxInputReportLength": 8,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 2
    },
```

```
    {
      "Time": "2021-12-19T23:35:49.7845394-05:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'HUION P608N'",
      "Level": 1
    },
    {
      "Time": "2021-12-19T23:35:49.7845485-05:00",
      "Group": "Plugin",
      "Message": "No constructor found for ''",
      "Level": 0
    },
    {
      "Time": "2021-12-19T23:35:49.7845572-05:00",
      "Group": "Detect",
      "Message": "Using device 'P608N'.",
      "Level": 0
    },
    {
      "Time": "2021-12-19T23:35:49.784566-05:00",
      "Group": "Detect",
      "Message": "Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.",
      "Level": 0
    },
    {
      "Time": "2021-12-19T23:35:49.7845753-05:00",
      "Group": "Detect",
      "Message": "Device path: \\\\?\\hid#vid_256c&pid_006e&mi_01&col02#8&276916e5&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Level": 0
    },
    {
      "Time": "2021-12-19T23:35:49.7845847-05:00",
      "Group": "Detect",
      "Message": "Found tablet 'HUION P608N'",
      "Level": 1
    },
    {
      "Time": "2021-12-19T23:35:49.7845995-05:00",
      "Group": "IOException",
      "Message": "Can't read from this device.",
      "Level": 3
    }
  ]
}
```

It should use huion's default drivers but it's having issues

jamesbt365, 2021-12-31:
help > export diagnostic, save that file send it here and also tablet > device string reader 9580 top box 110 in the middle and then dump all, also send that file here.

jxu, 2021-12-31:
I am probably the only person on Earth using this tablet so I just used the stock huion drivers

jxu, 2021-12-31:
I already posted the diagnostic interesting bits
[device_string_dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7795207/device_string_dump.txt)

Mrcubix, 2021-12-31:
i can tell you that you probably didn't, upload the entire device list

jamesbt365, 2021-12-31:
Still need the other parts of the diagnostic, still missing information we need.

jamesbt365, 2022-01-14:
We still need the entire diagnostics file, you can remove the envvars if you would like but we need everything else or we cannot create this configuration.

jxu, 2022-01-15:
I just used the default drivers

closed 2022-01-15 by jxu