OpenTabletDriver/OpenTabletDriver #1775: Parser is null when included in plugin

Mrcubix, 2021-12-20:
## Description

Using a parser provided by a plugin will result in a nullref on report, `Parse()` is not called.

# Example Parser

KappaCroPadReportParser:
```cs
using OpenTabletDriver.Plugin.Tablet;
using OpenTabletDriver.Plugin;

namespace TenMoon.Vendors
{
    public class KappaCroPadReportParser : IReportParser<IDeviceReport>
    {
        public IDeviceReport Parse(byte[] report)
        {
            return new KappaCroPadAuxReport(report);
        }
    }
}
```

KappaCroPadAuxParser:
```cs
using System;
using OpenTabletDriver.Plugin.Tablet;
using OpenTabletDriver.Plugin;
using TenMoon;

namespace TenMoon.Vendors
{
    public struct KappaCroPadAuxReport : IAuxReport
    {
        public KappaCroPadAuxReport(byte[] report)
        {
            Log.Debug("KappaCroPadAuxReport", $"{BitConverter.ToString(report)}");

            Raw = report;

            AuxButtons = new bool[]
            {
                report[1] != 0x0,
                report[2] != 0x0,
                report[3] != 0x0,
                report[4] != 0x0,
                report[5] != 0x0,
                report[6] != 0x0,
                report[7] != 0x0,
                report[8] != 0x0,
                report[9] != 0x0,
                report[10] != 0x0,
                report[11] != 0x0,
                report[12] != 0x0,
            };
        }

        public bool[] AuxButtons { set; get; }
        public byte[] Raw { set; get; }
    }
}
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 8.1 x64
| OpenTabletDriver Version | master (#1751)
| Tablet                   | PTK-640

# Diagnostic

```log
[Plugin:Debug]  Loading plugin '10moonParser'
[Plugin:Debug]  Loading plugin 'BezierInterpolator'
[Plugin:Debug]  Loading plugin 'RadialFollow'
[Detect:Info]   Searching for tablets...
[Detect:Debug]  Configuration overrides exist in 'C:\Users\samdra.r\Downloads\Op
enTabletDriver win-x64-master-#1751\Configurations', overriding built-in configu
rations.
[Detect:Debug]  Searching for tablet 'kappaCroPad'
[Device:Debug]  Initializing device 'KappacroPad' \\?\hid#vid_1c68&pid_02dd&mi_0
0#8&9e5bff5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Device:Debug]  Using report parser type 'TenMoon.Vendors.KappaCroReportParser'
[Detect:Info]   Found tablet 'kappaCroPad'
[Display:Debug] DPI Awareness enabled
[kappaCroPad:Info]      Output mode: Absolute Mode
[kappaCroPad:Info]      Display area: [3840x1080@<1920  540>:0°],
[kappaCroPad:Info]      Tablet area: [72,7x72,7@<36,35  36,35>:0°],
[kappaCroPad:Info]      Clipping: Enabled
[kappaCroPad:Info]      Ignoring reports outside area: Disabled
[kappaCroPad:Info]      Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info] Driver is enabled.
[NullReferenceException:Error]  Object reference not set to an instance of an ob
ject.
   at OpenTabletDriver.Devices.DeviceReader`1.Main() in D:\a\OpenTabletDriver\Op
enTabletDriver\OpenTabletDriver\Devices\DeviceReader.cs:line 106
[HidSharpDeviceRootHub:Debug]   Changes: 1, Add: 0, Remove: 1
[RootHub:Debug] Invoking DevicesChanged
```

Mrcubix, 2021-12-21:
After doing some more testing, building OTD with the parser included in `OpenTabletDriver.Configurations.Parser` does work.
This might confirm my guess that parsers located in `OpenTabletDriver.Plugin` as well as external parsers require some kind of provider

X9VoiD, 2021-12-21:
Not reproducible on latest master.

Mrcubix, 2021-12-21:
```cs
[Plugin:Debug]  Loading plugin '10moonParser'
[Plugin:Debug]  Loading plugin 'BezierInterpolator'
[Plugin:Debug]  Loading plugin 'RadialFollow'
[Detect:Info]   Searching for tablets...
[Detect:Debug]  Configuration overrides exist in 'C:\Users\samdra.r\Downloads\Pl
ugin-fix\Configurations', overriding built-in configurations.
[Detect:Debug]  Searching for tablet 'kappaCroPad'
[Device:Debug]  Initializing device 'KappacroPad' \\?\hid#vid_1c68&pid_02dd&mi_0
0#8&9e5bff5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Device:Debug]  Using report parser type 'TenMoon.Vendors.KappaCroReportParser'
[Detect:Info]   Found tablet 'kappaCroPad'
[Display:Debug] DPI Awareness enabled
```
it took me 12 seconds to check and it doesn't

closed 2021-12-21 by X9VoiD