OpenTabletDriver/OpenTabletDriver #1782: Support Request: plugins not working after Upgrading popOS

JonaLK, 2021-12-22:
# Support Request
Opentabletdriver used to work very well on my popOS System (Used to be Ubuntu 21.04), after upgrading to 21.10, plugins stopped working with the error message  
## What I've already tried
*reinstalling OpenTabletDriver
*reinstalling netframwork
*updating everything
*opening a bug report for the plugin 
https://github.com/AbstractQbit/AbstractOTDPlugins/issues/2
Which was stupid because other plugins are also affected. 

## Diagnostic Information
[Settings:Info]	Driver is auto-enabled.
[Plugin:Debug]	No constructor found for 'TabletDriverFilters.Devocub.Antichatter'
[Settings:Info]	Interpolator: TabletDriverFilters.Hawku.Smoothing
[DllNotFoundException:Error]	Unable to load shared library 'librt' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibrt: cannot open shared object file: No such file or directory

InfinityGhost, 2021-12-23:
This is a known issue, it's a unmarked dependency in our package. From a quick search with `apt-file` I was able to find a potential dependency, `libc6-dev` which should provide what's needed for interpolators.

JonaLK, 2021-12-23:
Thanks for the quick response! 
Unfortunately a apt-get install shows 
libc6-dev is already the newest version (2.34-0ubuntu3).
Which means it is installed already :(

InfinityGhost, 2021-12-23:
What's the output of `stat /usr/lib/librt.so*`? If it fails it may also be at `/lib/librt.so*`

JonaLK, 2021-12-24:
both result in the same:
stat: cannot statx '/usr/lib/librt.so*?': No such file or directory
and
bash: /lib/librt.so*: No such file or directory

InfinityGhost, 2021-12-24:
The only other thing that provides it is `libc6`, assuming that the v0.5.3.3 release doesn't rely on the headers. `LD_LIBRARY_PATH` might need updated to include `/usr/lib/x86_64-linux-gnu/` so that it can find that static library, but it really should not be required if the installation is correct.

JonaLK, 2021-12-24:
It is actually driving me nuts, sorry to bother on Christmas eve, but export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/x86_64-linux-gnu/
still did not fix it.
libc6 is on the newest version, uninstalling is (obviously) not possible since it would break the OS :/
It's not even an Opentabletdriver problem but rather from my OS...
Thank's for the help though :)

X9VoiD, 2021-12-24:
This is already fixed by #1592.

closed 2021-12-24 by X9VoiD

JonaLK, 2021-12-24:
How do I apply it though? :(

jamesbt365, 2021-12-24:
You download any build from this forward or patch it in yourself.