OpenTabletDriver/OpenTabletDriver #1785: Debugger tablet data size issues

Kuuuube, 2021-12-23:
## Description

When using tablets that send more than 4 rows of bytes in `Raw Tablet Data` or have more than 4 rows in `Tablet Report` the fifth line is halfway cut off and all lines past that point are not visible at all. 

Apparently on linux (and maybe macos), users can just expand the debugger to show these but that is not possible on windows.

Example:
[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Win10
| OpenTabletDriver Version | ed3cb17a3bfb0f960ccc9d6f18f273073b96d352

jamesbt365, 2021-12-23:
Does only a 5th row need to be shown?

Kuuuube, 2021-12-23:
No. There can be way more rows and it's hard to say how many but it would be best to not put a hard cap. 

The PTH-460 has 8 rows of raw data but it can do at least 20 touch inputs (though the parser is a bit broken right now) so it would need at least 20 rows.

jamesbt365, 2021-12-23:
Is there a way of making it expandable on windows?

gonX, 2021-12-24:
It's not great on Linux either, so resizing the tablet report (and raw data) field height dynamically would probably be preferable. If that isn't possible, I propose a minimum of 10 rows, and making it scrollable if it isn't already.

jamesbt365, 2021-12-24:
Dynamic sizing is probably the way to go, no idea how to do this though.

Kuuuube, 2021-12-24:
Dynamic sizing could get extremely annoying with multiple tablets plugged in. Many tablets send idle reports which would make the debugger window suddenly get larger or smaller then go back to the size of the current tablet.

gonX, 2021-12-24:
Tell me what you think of #1789

gonX, 2021-12-25:
Assuming based on your comments in the PR that it'll close this

closed 2021-12-26 by InfinityGhost