OpenTabletDriver/OpenTabletDriver #1794: Tablet not detecting

ChickenHumanPerson, 2021-12-25:
I have a Huion 420 and it is not detecting the tablet. I am not sure if it is the version I am using or not. I am using version 0.5.3.3 and something I did notice was this, (see image)
[image]
Please help me with this issue, I am completly lost.

ChickenHumanPerson, 2021-12-25:
I have figured out the issue. I listed the wrong tablet and realized that the tablet I have is actually the Huion H420. So, I must now wait until it is fixed.

closed 2021-12-25 by ChickenHumanPerson

Mrcubix, 2021-12-25:
[image]
- Pre-Release: https://cdn.discordapp.com/attachments/615939905566015493/922902857060081684/OpenTabletDriver.win-x64.zip
- Zadig : https://github.com/pbatard/libwdi/releases/download/b755/zadig-2.6.exe

- Create a new folder called `Configurations` and extract the content of this zip file inside