OpenTabletDriver/OpenTabletDriver #1797: Tablet not detected

frostishyper, 2021-12-25:
I Bought a new Veikk S640 yesterday its my first time using a tablet, I plug in my tablet and install OTB and the required framework for win 10 but when i open OTB it says no tablet detected

[frostishyper - diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7775567/frostishyper.-.diagnostics.txt)

jamesbt365, 2021-12-25:
Install this:
https://download.visualstudio.microsoft.com/download/pr/1daf85dc-291b-4bb8-812e-a0df5cdb6701/85455a4a851347de26e2901e043b81e1/windowsdesktop-runtime-5.0.12-win-x64.exe

jamesbt365, 2021-12-25:
Then follow:
https://mrcubix.github.io/OTD.Backport-Wiki/installation.html

frostishyper, 2021-12-25:
> Then follow:
> https://mrcubix.github.io/OTD.Backport-Wiki/installation.html

Thanks It works now

closed 2021-12-25 by frostishyper