OpenTabletDriver/OpenTabletDriver #1800: No Tablet Detected - Linux (AUR) Artist 12

DonArt-Q, 2021-12-25:
I have an Artist 12 that I want to use with OpenTabletDriver (since the official XP-Pen drivers aren't giving me much luck) however, when I open up OTD, I get the message 'connecting to OpenTabletDriver daemon' and 'No Tablet Detected'. Trying to detect a tablet with Ctrl+D only causes it to crash, and so does trying to export diagnostics.

jamesbt365, 2021-12-25:
Export diagnostic to a location otd actually has access to, like the desktop, home, documents, stuff like that.

DonArt-Q, 2021-12-25:
The issue with that is that pressing the Export Diagnostics button crashes before I am able to choose where to export the diagnostic file to.

gonX, 2022-01-18:
Try updating OpenTabletDriver to the 0.6.0.2 release and seeing if that works for you

DonArt-Q, 2022-01-30:
It still doesn't work, but I get an error message when trying to detect the tablet. Out of curiosity, does OTD need systemd on linux, since I'm using s6.
[image]

jamesbt365, 2022-01-30:
You need to start the daemon and no, otd doesn't need systemd.

DonArt-Q, 2022-01-30:
How do I start the daemon?

DonArt-Q, 2022-01-30:
nevermind, running otd in the terminal worked

jamesbt365, 2022-01-30:
if you used a package manager `otd` will just work, otherwise you will need to specify the path.

DonArt-Q, 2022-01-30:
my tablet still isn't detected though.

jamesbt365, 2022-01-30:
Now export a diagnostic, you are probably missing udev.

DonArt-Q, 2022-01-30:
Here's the diagnostic info:
[diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7966343/diagnostic.txt)

DonArt-Q, 2022-01-30:
*sigh*

The issue was that I was supposed to connect the tablet after the daemon was started... Thanks for helping me with this though.

closed 2022-01-30 by DonArt-Q