OpenTabletDriver/OpenTabletDriver #1801: Add support for Trust Flex Design Tablet

Cyanoure, 2021-12-26:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    |Trust
| Product Name    |Flex
| Width           |5.5 "
| Height          |4 "
| Pressure Levels |1024
| LPI             |2500

## Diagnostic Information

```
    {
      "DevicePath": "\\\\?\\hid#vid_2179&pid_0077&mi_00&col03#8&2e812bfb&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 8569,
      "ProductID": 119,
      "CanOpen": true,
      "ReleaseNumberBcd": 0,
      "ReleaseNumber": "0.0",
      "ProductVersion": 0,
      "Manufacturer": "UGTABLET",
      "ProductName": "TABLET GT5040",
      "SerialNumber": "",
      "MaxInputReportLength": 0,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 8
    },
    {
      "DevicePath": "\\\\?\\hid#vid_2179&pid_0077&mi_00&col02#8&2e812bfb&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 8569,
      "ProductID": 119,
      "CanOpen": false,
      "ReleaseNumberBcd": 0,
      "ReleaseNumber": "0.0",
      "ProductVersion": 0,
      "Manufacturer": "UGTABLET",
      "ProductName": "TABLET GT5040",
      "SerialNumber": "",
      "MaxInputReportLength": 8,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "\\\\?\\hid#vid_2179&pid_0077&mi_00&col01#8&2e812bfb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 8569,
      "ProductID": 119,
      "CanOpen": false,
      "ReleaseNumberBcd": 0,
      "ReleaseNumber": "0.0",
      "ProductVersion": 0,
      "Manufacturer": "UGTABLET",
      "ProductName": "TABLET GT5040",
      "SerialNumber": "",
      "MaxInputReportLength": 6,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
    {
      "DevicePath": "\\\\?\\hid#vid_2179&pid_0077&mi_01&col02#8&aa9ee39&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 8569,
      "ProductID": 119,
      "CanOpen": true,
      "ReleaseNumberBcd": 0,
      "ReleaseNumber": "0.0",
      "ProductVersion": 0,
      "Manufacturer": "UGTABLET",
      "ProductName": "TABLET GT5040",
      "SerialNumber": "",
      "MaxInputReportLength": 0,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 3
    }
```

Interesting, it's not shown as 'Trust Flex'. It's just 'Pen Tablet' in the Device Manager.
[image]

Website: https://www.trust.com/en/product/21259-flex-design-tablet-black

Mrcubix, 2021-12-26:
with manufacturer drivers up and running:

- https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool

- When done, head to `C:\Program Files\USBPcap` in the explorer (you can probably copy paste that path in the path bar)

- Once you're there, start `USBPcapCMD.exe`

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to `C:\Program  Files\USBPcap`,

- Start `USBPcapCMD.exe`,

- screenshot the console.

Cyanoure, 2021-12-26:
[image]
[image]
Sorry for Hungarian Windows.

Mrcubix, 2021-12-26:
- type 2 then enter,
- name the file. Press enter and do the following steps:

- Move for 5s
- Unplug the tablet from your computer (remove the usb cable from the port),
- Wait 2 seconds,
- Plug in your tablet again (connect your tablet to your pc ),
- Wait 2 seconds,
- Move for 5s 
- Close the console (red cross),
- A new file has been generated with the name you gave in the same folder,
- Upload that file here

Cyanoure, 2021-12-26:
[trustflext_pcap.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7776560/trustflext_pcap.zip)

Mrcubix, 2022-01-13:
Right click on one of the devices above > update > Update manually > Let me pick from a list > Screenshot the device available  to choose from there

Cyanoure, 2022-01-17:
What do you mean? This?
[image]

Cyanoure, 2022-01-17:
And if I uncheck the checkbox:
[image]
[image]
[image]
[image]
[image]
[image]

Kaleidosium, 2022-10-27:
+1 For this, The official drivers don't work on my sister's laptop for whatever reason, hopefully OTD works better.

jamesbt365, 2024-03-26:
It has been a while, if this is still in demand uninstall the OEM driver and upload a new diagnostic from within OTD. Thanks!

closed 2025-10-06 by gonX