OpenTabletDriver/OpenTabletDriver #1811: daemon crash on launch due updater?

OniCado, 2021-12-27:
## Description

daemon exits immediately after launching
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Win 10 Pro N 21H2 Build 19044.1415
| OpenTabletDriver Version |  d758023
| Tablet                   | ---

OniCado, 2021-12-27:
[daemon.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7780832/daemon.log)

jamesbt365, 2021-12-27:
Can repoduce on a windows virtual machine on d758023

jamesbt365, 2021-12-27:
Is this an issue with macos or just windows?

Manson-22, 2021-12-28:
im on an older version most likely and it works fine. just installed on a new computer and having same issue.

gonX, 2021-12-28:
The stacktrace is in `WindowsUpdater.cs` so I'm assuming this is Windows-only for now.

vedattt, 2021-12-28:
Inner stack trace
[image]

jamesbt365, 2021-12-28:
Is this a problem specifically always with the latest master trying to do something?

vedattt, 2021-12-28:
> Is this a problem specifically always with the latest master trying to do something?

Yes, it is, and on both MacOS and Windows most likely. Will push a PR to fix it in a few moments though.

PR #1817 (2022-01-02): Fix `Updater` crashing daemon due to revision id
## Changes

- `InformationalVersion` is for non-programmatic uses, and is not safe to build into a `Version` object. The `Version` instance provided by assemblies is better for the purposes here.

## Issues

- Closes #1811
closed 2022-01-02 by InfinityGhost