OpenTabletDriver/OpenTabletDriver #1813: no tablet found (wacom intuos)

Mayochano, 2021-12-27:
hi, Wacom Intuos Pen & Touch is not being detected at all on the console and no where to be seen.

Operating system : PopOS 21.04
Tablet: Wacom Intuos Pen & Touch 2015 version

jamesbt365, 2021-12-27:
There is about 40 wacom intuos tablets, so that literally does not narrow it down, model number off the back and

Help > export diagostic save that file as a txt and send it here, it most probably is part of the faq

Mayochano, 2021-12-27:
okay i've checked the back of my tablet. my tablet model is Wacom CTH-490 but it didn't detect.

jamesbt365, 2021-12-27:
You have still ignored like 70% of my message

Mayochano, 2021-12-27:
> 
damn, chill lol, don't have to be rude smh. 
[tablet.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7781531/tablet.txt)

Mrcubix, 2021-12-27:
[image]

https://opentabletdriver.net/Wiki/FAQ/Linux#fail-device-streams

Mayochano, 2021-12-27:
> 

i've tried that, maybe I did it wrong.

Mayochano, 2021-12-27:
let me try again

Mayochano, 2021-12-27:
lana@pop-os:~/OpenTabletDriver$ ./generate-rules.sh
/usr/share/dotnet/sdk/5.0.404/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(141,5): error NETSDK1045: The current .NET SDK does not support targeting .NET 6.0.  Either target .NET 5.0 or lower, or use a version of the .NET SDK that supports .NET 6.0. [/home/lana/OpenTabletDriver/OpenTabletDriver.Tools.udev/OpenTabletDriver.Tools.udev.csproj]

The build failed. Fix the build errors and run again.

Mrcubix, 2021-12-27:
https://github.com/OpenTabletDriver/OpenTabletDriver/suites/4765323643/artifacts/132500511

skip the build part and move this file to the path in the faq

jamesbt365, 2021-12-27:
Either grab the latest udev from the actions builds on master or download the dotnet 6 sdk

Mayochano, 2021-12-27:
it works now thxs

closed 2021-12-27 by Mayochano