OpenTabletDriver/OpenTabletDriver #1815: Use native timer on Linux

gonX, 2021-12-28:
## Description
After #1802 was merged we are no longer using native timers on Linux.

Should they get reimplemented its important that the memory leak from #1727 is fixed.

Setting this as low priority for now as there doesn't seem to be any noticeable performance degradation on my system

25pwn, 2022-01-12:
Can you add a patch that doesn't use the timer while there is no input?

gonX, 2022-01-15:
Increasing to medium priority as its probably fairly catastrophic for low core count users. 1 core is pegged at 100% for me

AkiSakurai, 2022-02-01:
It is likely caused by regression introduced by doenet 6.0.
https://github.com/dotnet/runtime/issues/64509#issuecomment-1025140207

jamesbt365, 2022-06-05:
#2190

closed 2022-06-05 by jamesbt365