OpenTabletDriver/OpenTabletDriver #1816: Daemon needs to be restarted when tablet is replugged

terencode, 2021-12-28:
## Description
If the tablet is disconnected (computer is suspended, tablet screen is turned off, usb is unplugged...), I can't use it until I restart the opentabletdriver service. 
What is weird is that the daemon sees the tablet reappeared and says it's applying the different settings I configured earlier, but it does nothing. Same behaviour if I manually trigger it (pressing 'Apply' from the GUI).

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | ArchLinux
| OpenTabletDriver Version |d7580231
| Tablet                   | Kamvas pro 20

gonX, 2022-01-05:
Can't reproduce by replugging on ab62eceb - both with absolute and artist mode.

Pen passthrough was removed recently, btw.

Try resetting your configuration (delete or move `~/.config/OpenTabletDriver`) and updating your install to a recent commit.

gonX, 2022-01-18:
Is there a chance this is the same as #1870 ?

terencode, 2022-01-18:
Looks like it. I have found that pressing the "redetect tablets" was enough to fix it as you suggested on discord. 
However, I'm not sure if it worked when just reapplying settings as suggested in that issue.