OpenTabletDriver/OpenTabletDriver #1821: Multikey bindings doesn't respect keyboard layout

terencode, 2021-12-28:
## Description
My keyboard layout is AZERTY and when I try to set ctrl+a, I get ctrl+q instead.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | ArchLinux
| OpenTabletDriver Version |f06a0ed60947bede5413dc67e89f9eaf0d263b74
| Tablet                   | Kamvas pro 20

jamesbt365, 2021-12-28:
this doesn't seem to affect linux, i'm using `setxkbmap` to change my keyboard layout to dvorak and it behaves properly

terencode, 2021-12-29:
Sorry, I did not recall correctly what happened.
On the interface, the keys are correctly displayed so when I press ctrl + a I get ctrl + a.
However, when I press the button on the tablet, I don't get ctrl+a but ctrl+q. In order to get ctrl+a I have to bind it to ctrl+q.

Quantenzitrone, 2022-06-24:
Why is this low priority?

This is totally confusing, as set shortcuts just don't work until you realize how the keys get typed.

Quantenzitrone, 2022-06-24:
A way to fix this, with a bit more confusion than respecting keyboard layouts, but I think also less work, would be converting everything into querty.
So typing the first key in the first letter row when setting the shortcut keys would set Q and type Q afterwards etc..
In general, showing the keys that actually get typed would make less confusion.

jamesbt365, 2022-06-24:
> Why is this low priority?
>
> This is totally confusing, as set shortcuts just don't work until you realize how the keys get typed.

This is currently marked as low priority due to higher priority issues such as core features entirely not working and mostly because we are waiting on the dependency injection rewrite to be completed before any work goes on those. This is more of an inconvenience anyway but i'll look into it sometime soon myself.

gonX, 2022-06-24:
It's not low priority per se, it's planned for the next larger release (if any) before the big rewrite for 0.7 as @jamesbt365 mentions. We (or at least I personally) use priority tags to indicate which issues are best to work on first, so if there are no high or medium priority issues we can work on any of the low priority tags.

But thanks for pointing out that we likely need to better define what priority tags mean. I agree this issue a bit more than low priority, but I believe at the time of tagging we had larger issues at hand.

m10d, 2023-09-18:
@jamesbt365 the issue I submitted #2932 does seem likely to share an underlying root cause. 

In my case, there are some additional elements that i want to ensure are captured here - specifically the OTD  gui app is doing two different things with the keys:
1. it's "capturing" and displaying the correct / remapped key (I press `physical_capslock`, the UI displaces `ctrl` **as i would want**)
2. but it's sending something **different**, which does not match what it says it's going to send.

Do you agree, or is there something not unique about that aspect?

jamesbt365, 2023-09-18:
As you can see from this issue, it is the same issue, as the GUI shows the right key.

jamesbt365, 2023-09-18:
Please look for duplicates and read the relevant issues next time, thanks!

m10d, 2023-09-18:
> Please look for duplicates and read the relevant issues next time, thanks!

fair enough - I did look through the issues, and saw this (albeit missed this in the thread where what you point out was determined).  

There is a unique treatment of capslock in my issue - **I cannot press a capslock combination** (at all) in the UI.  I had to observe and manually duplicate the exact way OTD phrases the keycode, and hope that it's parsed ~symmetrically.

The user in this issue has a workaround, in that in the UI they can press the pre-mapped key and OTD at least picks it up there.  I'm fine with these bugs being merged - but this is different and it makes my use case more difficult for the user to decipher.  Perhaps this doesn't change the ultimate priority, but it does broaden the impact.

jamesbt365, 2023-09-18:
> There is a unique treatment of capslock in my issue - I cannot press a capslock combination (at all) in the UI.

eto (external) bug.

X9VoiD, 2023-09-18:
OTD GUI receives translated keys, OTD Daemon sends out raw keycode.

Hufschmidt, 2025-03-10:
> It's not low priority per se, it's planned for the next larger release (if any) before the big rewrite for 0.7 as [@jamesbt365](https://github.com/jamesbt365) mentions. We (or at least I personally) [...]

It would be cool if this could be added to the FAQ / Known Issues on the website.
I stumbled upon this today when trying to get my keybinds working and initially thought that no tablet key press was registering when in fact opentabletdriver was sending CTRL+Y instead of CTRL+Z the whole time. (German/DE keyboard layout here)

gonX, 2026-09-25:
Putting this on the backlog instead of the Avalonia milestone so that it's at least known about, but fixing this is extremely low priority given that workarounds exists, and there is no clear path forward for fixing this. I assume that most GUI frameworks do not support receiving raw key codes.

edit: see OpenTabletDriver/opentabletdriver.github.io/pull/285 for documentation