OpenTabletDriver/OpenTabletDriver #1824: CTH-460 pen doesn't work

SuperKKPL, 2021-12-29:
I connected the tablet to the computer and the tablet stylus does not work, only the touch function.

jamesbt365, 2021-12-29:
can you go to help > export diagnostics, save that file as a txt and send it here please?

SuperKKPL, 2021-12-29:
[123.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7788957/123.txt)

jamesbt365, 2021-12-29:
You have wacom drivers installed, you need to uninstall them through apps and features, if they are not there they were uninstall incorrectly and therefore you need to install them then uninstall through apps and features.

(also oem kill doesn't really do anything anymore)

SuperKKPL, 2021-12-29:
ok, I'll try

SuperKKPL, 2021-12-29:
I tried uninstalling every driver and still noting.
But I tried connecting the CTH-480 and it works, but I really want to play on CTH-460.
Can I do something else to the drivers?

jamesbt365, 2021-12-29:
Restart otd and export another diag.

SuperKKPL, 2021-12-29:
[4562.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7789174/4562.txt)

jamesbt365, 2021-12-29:
close otd, does your tablet work?

SuperKKPL, 2021-12-29:
When i close OTD my tablet does not work, only touch function work.

SuperKKPL, 2021-12-29:
I also tried connecting tablet to the other PC which never before had tablet drivers and it's the same.

jamesbt365, 2021-12-29:
Try this: https://mrcubix.github.io/OTD.Backport-Wiki/installation.html

SuperKKPL, 2021-12-29:
It does not work too.

jamesbt365, 2021-12-29:
remove the backport plugin then replace the config with this:
[AAAAAAAA.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790529/AAAAAAAA.zip)

SuperKKPL, 2021-12-29:
It does work! Thanks mate!

closed 2021-12-29 by jamesbt365