OpenTabletDriver/OpenTabletDriver #1827: Add support for Huion Kamvas 16 2021 (GS1562)

Casul51, 2021-12-29:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | Huion 
| Product Name    | Kamvas 16(2021)
| Width           | 1920
| Height          | 1080
| Pressure Levels  | 8192
| LPI             | 5080

## Diagnostic Information

[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7789910/diag.txt)

## Huion dump
[huion.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7789921/huion.txt)

Also worth to mention, without this driver I had pen as usb device.
System: Arch linux.

jamesbt365, 2021-12-29:
are you on opentabletdriver-git?

jamesbt365, 2021-12-29:
Put this config in usr/share/OpenTabletDriver/Configurations/Huion
[aaaaaaaa.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7789984/aaaaaaaa.zip)

Casul51, 2021-12-29:
> are you on opentabletdriver-git?

yes

jamesbt365, 2021-12-29:
thats fine, just use the config above, you will have to make the config folder.

Casul51, 2021-12-29:
> Put this config in usr/share/OpenTabletDriver/Configurations/Huion, make sure to extract the zip [aaaaaaaa.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7789984/aaaaaaaa.zip)

Yep, pen and all buttons works. But pen's pressure and tilt don't.

jamesbt365, 2021-12-29:
Pressure should be able to be used with artist mode, but lets make sure everything else works, can you get the max x and y from the tablet > tablet debugger and then move your pen slowly to the bottom right of your tablet.

Casul51, 2021-12-29:
> Pressure should be able to be used with artist mode, but lets make sure everything else works, can you get the max x and y from the tablet > tablet debugger and then move your pen slowly to the bottom right of your tablet.

In artist mode pressure and tilt work. Also it seems I have entire screen space working. But in artist mode pen's buttons don't work.

gonX, 2021-12-29:
You have to bind pen barrel buttons manually for now

Casul51, 2021-12-29:
It might be because right now I'm in X11, since my AMD GPU in service center.

jamesbt365, 2021-12-29:
Does the tilt show up in the debugger?

Casul51, 2021-12-29:
Yes, tilt works

jamesbt365, 2021-12-29:
> Pressure should be able to be used with artist mode, but lets make sure everything else works, can you get the max x and y from the tablet > tablet debugger and then move your pen slowly to the bottom right of your tablet.

can you do this quickly

Casul51, 2021-12-29:
> You have to bind pen barrel buttons manually for now

Manually? Where, in pen settings? I've already bind them.

Casul51, 2021-12-29:
> > Pressure should be able to be used with artist mode, but lets make sure everything else works, can you get the max x and y from the tablet > tablet debugger and then move your pen slowly to the bottom right of your tablet.
> 
> can you do this quickly?

Actually no. There _probably_ big delay and pen don't reach lower right point when in out of tablet.

gonX, 2021-12-29:
Just going based off this:

> But in artist mode pen's buttons don't work.

It needs to look similar to this:
[image]

jamesbt365, 2021-12-29:
get the max x and y then, going to the bottom right of your tablet, ignore everything else but the position, go the bottom right of your tablet slowly and get the max x and y

Casul51, 2021-12-29:
> Just going based off this:
> 
> > But in artist mode pen's buttons don't work.
> 
> It needs to look similar to this: [image]
> 
> /e: this is btw unrelated to area

Oh, I see, that works.

Casul51, 2021-12-29:
> get the max x and y then, going to the bottom right of your tablet, ignore everything else but the position, go the bottom right of your tablet slowly and get the max x and y

68840,38720

Casul51, 2021-12-29:
> > can you do this quickly?
> 
> Actually no. There _probably_ big delay and pen don't reach lower right point when in out of tablet.

Maybe I'm just too fast? 
I actually didn't even needed to make such fast strokes before.

jamesbt365, 2021-12-29:
does all the core features work?

Casul51, 2021-12-29:
Actually all features work!

And I do believe that it is me who is moving stylus too fast. If I don't try to stroke entirely out of tablet, there is gap out of screen where stylus still register, so I guess it is normal. And drawing feels like usual.

You guys made a good job! Support was awesome!

closed 2021-12-29 by Casul51