OpenTabletDriver/OpenTabletDriver #1828: Support Request: Sorry to be annoying, but XP-Pen Deco fun S isn't being detected

Zerimm, 2021-12-29:
I recently changed tablet from XP-Pen g640 to this one, and it just isn't being detected.
I've tried reinstalling both the driver and .NET but it still isn't.

Here is diagnostic data
[Diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790011/Diagnostic.txt)

Sublimelime, 2021-12-29:
Did OTD generate the diagnostic like that, or did you strip the json formatting from the file? It's okay to upload it as txt, but the json formatting (all the brackets, the quotation marks, the indentation, etc) makes it easier to read.

jamesbt365, 2021-12-29:
upload another diag, this time do not remove the json formatting, just rename it to a txt.

Zerimm, 2021-12-29:
here
[Diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790056/Diagnostic.txt)

jamesbt365, 2021-12-29:
you will need to do this:
https://mrcubix.github.io/OTD.Backport-Wiki/installation.html

Zerimm, 2021-12-29:
Sorry, but that still doesn't work

Mrcubix, 2021-12-29:
export another diag

Mrcubix, 2021-12-29:
nvm, this is a new variation, extract the content of this zip file in `Configurations/XP-Pen`, replacing the old config

[CT430.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790095/CT430.zip)

Zerimm, 2021-12-29:
Do I need to do anything else? Sorry but it still doesn't work

Mrcubix, 2021-12-29:
export a diag and upload it here

Zerimm, 2021-12-29:
[Diagnosjhos.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790113/Diagnosjhos.txt)

Mrcubix, 2021-12-29:
well i can definitely tell you that the new config hasn't been loaded, either try restarting or re-extract the config properly

Mrcubix, 2021-12-29:
[CT640.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790136/CT640.zip)
nvm, XP-Pen naming scheme suck ass, Should be S, M & L, not XS, S & L

Zerimm, 2021-12-29:
It works now, thank you

closed 2021-12-29 by Zerimm