OpenTabletDriver/OpenTabletDriver #1829: Huion Kamvas 16 2021 GS1562 Object reference not set to an instance of an object.

dindybutts, 2021-12-29:
## Description

Hi, I just found this project today and wanted to try it with my huion tablet but it seems to not be detected, After poking around the issues here I came across [this](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1827) similar issue.

When using the provided config from that issue it seems to be detected now but throws the following when the pen is anywhere near the tablet
`[NullReferenceException:Error]	Object reference not set to an instance of an object.`

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 11 |
| OpenTabletDriver Version | 0.5.3.3 |
| Tablet                | Huion Kamvas 16 2021 (GS1562) |

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790292/diagnostics.txt)

dindybutts, 2021-12-29:
I should mention that the official drivers have been uninstalled completely

jamesbt365, 2021-12-29:
This is a master config, i will convert this to release so it works.

jamesbt365, 2021-12-29:
[aaaa.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7790330/aaaa.zip)

dindybutts, 2021-12-29:
Gotcha, the config stopped it from throwing that error and everything other than pressure seems to be working fine (pressure stuff might be on my end though), Would it be safe to assume that master config will be included in the next release by default?

jamesbt365, 2021-12-29:
You need to use these go get pressure.

https://github.com/X9VoiD/vmulti-bin/releases/tag/v1.0
https://github.com/X9VoiD/VoiDPlugins/wiki/WindowsInk

And yes, this config should be included in next release

closed 2021-12-29 by jamesbt365

dindybutts, 2021-12-29:
Perfect, thanks for the help!