OpenTabletDriver/OpenTabletDriver #1831: Huion H420X Not Detected

OblivionGhoul, 2021-12-29:
## Description
I have a Huion Inspiroy H420X and it is not being detected. I already tried making a custom config in #1264, but it is still not detecting my tablet. 

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10 Pro
| OpenTabletDriver Version | v0.5.3.3
| Tablet                   | Huion Inspiroy H420X

jamesbt365, 2021-12-29:
help > export diagnostic, save that file as a txt and send it here

OblivionGhoul, 2021-12-30:
[text.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7794612/text.txt)

Mrcubix, 2021-12-30:
https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

OblivionGhoul, 2021-12-30:
I just tried running that script. I know that it worked because when I ran the script again, it said "No Gaomon/Huion driver detected." But it is still not detecting my tablet. Maybe because my config file is incorrect, but I got it from #1264?

The Logs: [logs.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7794671/logs.txt)

Mrcubix, 2021-12-30:
uninstall razer synapse if installed then right click on start menu > Device manager >View > Device by container > look for Gaomon Tablet there > open it's container (arrow on the left) > screenshot

OblivionGhoul, 2021-12-30:
[image]

Mrcubix, 2021-12-30:
[image]

right click on this one > Update > Second option > Let me pick from a list > pick `HID compliant device`

OblivionGhoul, 2021-12-30:
[image]
It was `USB Input Device`, so I installed the drivers and it looks like this now.

Mrcubix, 2021-12-30:
alright now https://mrcubix.github.io/OTD.Backport-Wiki/installation.html

OblivionGhoul, 2021-12-31:
Yes, it can now detect my tablet. Thank you!

closed 2021-12-31 by OblivionGhoul