OpenTabletDriver/OpenTabletDriver #1839: NRE on latest master (d758023) when starting the otd service

jkcdarunday, 2021-12-30:
## Description
I get the following error from `journalctl --user -u opentabletdriver` when I starting the `otd` service very similar to what happens in #1703. That one is supposedly already fixed though so this could be a resurfacing of the same bug or something.
```
Dec 31 07:28:02 [redacted] otd[47782]: [Detect:Debug]        Searching for tablet 'XP-Pen Star G640'
Dec 31 07:28:02 [redacted] otd[47782]: [Detect:Debug]        Searching for tablet 'XP-Pen Star G640S'
Dec 31 07:28:02 [redacted] otd[47782]: [Detect:Debug]        Searching for tablet 'XP-Pen Star G960'
Dec 31 07:28:02 [redacted] otd[47782]: [Detect:Debug]        Searching for tablet 'XP-Pen Star G960S Plus'
Dec 31 07:28:02 [redacted] otd[47782]: [Detect:Debug]        Searching for tablet 'XP-Pen Star G960S'
Dec 31 07:28:02 [redacted] otd[47782]: [Detect:Debug]        Searching for tablet 'Huion RTP-700'
Dec 31 07:28:02 [redacted] otd[47782]: [Device:Debug]        Initializing device 'Huion Tablet_H1162' /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:256C:006D.0002/hidraw/hidraw1
Dec 31 07:28:02 [redacted] otd[47782]: [Device:Debug]        Using report parser type 'OpenTabletDriver.Vendors.UCLogic.UCLogicReportParser'
Dec 31 07:28:02 [redacted] otd[47782]: [Device:Debug]        Initialized string index 200
Dec 31 07:28:02 [redacted] otd[47782]: [Detect:Info]        Found tablet 'Huion RTP-700'
Dec 31 07:28:02 [redacted] otd[47782]: [IPC:Debug]        Connected to a client.
Dec 31 07:28:02 [redacted] otd[47782]: Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Profiles.AreaSettings.GetDefaults(DigitizerSpecifications digitizer) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/AreaSettings.cs:line 70
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Profiles.AbsoluteModeSettings.GetDefaults(DigitizerSpecifications digitizer) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/AbsoluteModeSettings.cs:line 51
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Profiles.Profile.GetDefaults(TabletReference tablet) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/Profile.cs:line 61
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Profiles.ProfileCollection.<>c.<.ctor>b__2_0(TabletReference s) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/ProfileCollection.cs:line 22
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Linq.Utilities.<>c__DisplayClass2_0`3.<CombineSelectors>b__0(TSource x)
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Linq.Enumerable.SelectIListIterator`2.MoveNext()
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Collections.ObjectModel.ObservableCollection`1.CreateCopy(IEnumerable`1 collection, String paramName)
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Collections.ObjectModel.ObservableCollection`1..ctor(IEnumerable`1 collection)
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 profiles) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/ProfileCollection.cs:line 17
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 tablets) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Profiles/ProfileCollection.cs:line 22
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Settings.GetDefaultProfiles() in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Settings.cs:line 59
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Settings.GetDefaults() in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Settings.cs:line 49
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Migration.SettingsMigrator.Convert(Settings old) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Migration/SettingsMigrator.cs:line 36
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Migration.SettingsMigrator.Migrate(FileInfo file) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Migration/SettingsMigrator.cs:line 31
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Desktop.Migration.SettingsMigrator.Migrate(AppInfo appInfo) in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Desktop/Migration/SettingsMigrator.cs:line 21
Dec 31 07:28:02 [redacted] otd[47782]:    at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings() in /home/[redacted]/.cache/paru/clone/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.Daemon/DriverDaemon.cs:line 213
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__127_1(Object state)
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Threading.ThreadPoolWorkQueue.Dispatch()
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
Dec 31 07:28:02 [redacted] otd[47782]:    at System.Threading.Thread.StartCallback()

```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Arch Linux |
| OpenTabletDriver Version | Build of latest master branch (d7580231) through the opentabletdriver-git AUR package |
| Tablet                   | Huion RTP-700 with config from #1445 ||

Mrcubix, 2021-12-30:
This could be due to an attempt at loading a `settings.json` file from stable (0.5.3.3).

As far as i'm aware, the format changed, so you would need to delete it.

I don't remember the path for it myself so i'll expecting someone to indicate the path here

jkcdarunday, 2021-12-30:
Hmm. I saw settings.json in `~/.config/OpenTabletDriver` and deleted the whole directory then restarted the otd server. Unfortunately, the same error occurs. Could it be related to the RTP-700 file that I'm using?

Mrcubix, 2021-12-30:
see the edit i've made

jkcdarunday, 2021-12-31:
Awesome, thanks! I've ported my config file to the new format and things seem to be working and I'm not getting the error anymore (initially forgot to changed the namespace but eventually resolved it).

Here's what I ended up with for other people's reference:
[RTP-700.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7794954/RTP-700.zip)

I will close this ticket now.

closed 2021-12-31 by jkcdarunday