OpenTabletDriver/OpenTabletDriver #1845: Double cursor

etherealxyz, 2022-01-01:
## Description

After shrinking the area, two cursors appeared. One works for the entire area, the other for the part that I have selected.
file with diagnostics attached
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System   |Win 11 |
| OpenTabletDriver Version  |v.0.5.3.3 |
| Tablet     |Xp-pen Star 06C |
[123.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7797679/123.txt)

jamesbt365, 2022-01-01:
This is because you have other drivers installed.

etherealxyz, 2022-01-01:
> 

did not install

Mrcubix, 2022-01-01:
The only ways you can get a double cursor in stable which is apparently what you're using are:

- Using multiple Interpolators at the same time (You should only be using 1, you mean can add as much filters as you want on top)
- Having another driver Installed and running (May not show up in the control pannel or Windows settings),
in which case you'll either Have to:
    - Find and use the uninstaller present in XP-pen Drivers files
    - Try to install them again and uninstall them and see if it make a diff

closed 2022-01-01 by Mrcubix