OpenTabletDriver/OpenTabletDriver #1848: Support Request: Support for Huion Kamvas Pro 16 is listed according to the docs but doesn't appear in the menu

top-kat, 2022-01-01:
# Support Request

## What I've already tried

Hi, Thanks for such a good project :star_struck: 

I have a Huion Kamvas pro 16 that is not recognized. Also, althougth it appear in the doc that lists compatible drivers, it doesn't appear in the application `configuration editor` not in the console listing when trying to detect the tablet...

Any help would be greatly appreciated :)

[diagtablet.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7798118/diagtablet.txt)

## Diagnostic Information

jamesbt365, 2022-01-01:
[Uploading pro16.zip…]()
Put this in Configurations\Huion

jamesbt365, 2022-01-02:
Assuming is resolved.

closed 2022-01-02 by jamesbt365

top-kat, 2022-01-02:
Hi I tried everything but still doesn't work
I didn't find the configuration file, I tried to create configurations in `~/.config` an also the downloaded folder of the app but the tablet config file were not recognized..
I also tried to add it manually, but whatever I do, when I save the configuration the app crashes.
Also tablet is not detected in tablet_debugger nor when I try to generate configuration. The only way it is recognized by the system is when I install digimend drivers. But in this case it is still not recognized by openTabletDriver...

Have you got any clue on how I can get forward ?

jamesbt365, 2022-01-02:
The configurations folder is in /usr/share/OpenTabletDriver/Configurations