OpenTabletDriver/OpenTabletDriver #1849: Location of the systemd service file for use as reference for creating a runit start script

Megaqualle, 2022-01-02:
I would like to use this project on my Linux computer, but, as I use Artix-Linux, I don't use systemd, but chose to use runit instead. This means that I have to write my own runit start script. As I am not an expert regarding that, I would like to use the service file as reference, but have been unable to find it.
This is what my script contains so far, which doesn't work:
```
#! /bin/sh

exec 2>1
exec /usr/bin/otd
```

hwsmm, 2022-01-02:
A service file of OTD (`opentabletdriver.service`) should be located in `/usr/lib/systemd/user`.

I don't know how runit works, but I don't recommend using any init systems that don't support user services properly since OTD doesn't really work well with root. You'd better use something like XDG Autostart or let your DE/WM start OTD after logging in.

Megaqualle, 2022-01-02:
I did not think of just starting it with my window manager, that is a much better idea.
Thanks for the tip.

closed 2022-01-02 by Megaqualle