OpenTabletDriver/OpenTabletDriver #1852: Display tablet calibration

gonX, 2022-01-02:
## Description
Some display tablets seem to have tablet active area larger than display dimensions. This results in some hard-to-fix area distortion that makes those screen tablets fairly undesirable to use with OTD.

Therefore I am proposing that OTD gets display tablet calibration. We likely just need the standard "tap these 5 crosshairs" calibration tool.

gonX, 2022-01-18:
Betting on a risky 0.7.x milestone on this one, though I'm totally open to pushing it back again. Until this is implemented, some display tablets are useless for drawing on OTD.

Koterminion, 2022-08-11:
I am in agreement of this since there are some callibration issues with the Cintiq 16. Jamesbt365 configured the support for the cintiq 16 but it seems like it still has offset issues.
[Here is the JSON file they managed to create out of my Cintiq 16.](https://pastebin.com/adWudPqX)