OpenTabletDriver/OpenTabletDriver #1855: Daemon crashes if standard output is unavailable

InfinityGhost, 2022-01-02:
## Description

If there is no output stream, the daemon will crash when logging.

```
Application: OpenTabletDriver.Daemon.exe
CoreCLR Version: 6.0.121.56705
.NET Version: 6.0.1
Description: The process was terminated due to an unhandled exception.
Exception Info: System.IO.IOException: No process is on the other end of the pipe.
   at System.ConsolePal.WindowsConsoleStream.Write(ReadOnlySpan`1 buffer)
   at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)
   at System.IO.StreamWriter.WriteLine(String value)
   at System.IO.TextWriter.SyncTextWriter.WriteLine(String value)
   at System.Console.WriteLine(String value)
   at OpenTabletDriver.Daemon.DriverDaemon.<.ctor>b__0_0(Object sender, LogMessage message)
   at OpenTabletDriver.Plugin.Log.Write(String group, String text, LogLevel level, Boolean createStackTrace, Boolean notify)
   at OpenTabletDriver.Plugin.Log.Debug(String group, String text)
   at OpenTabletDriver.Devices.RootHub.HandleHubDeviceNotification(Object sender, DevicesChangedEventArgs eventArgs)
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpDeviceRootHub.<.ctor>b__0_0(Object sender, DeviceListChangedEventArgs e)
   at HidSharp.DeviceList.RaiseChanged()
   at HidSharp.Platform.Windows.WinHidManager.DeviceMonitorEventThread()
   at System.Threading.Thread.StartCallback()
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| OpenTabletDriver Version | v0.6.0rc1

gonX, 2022-01-03:
Are we building Windows without debug symbols? Kinda annoying

gonX, 2022-01-03:
```diff
diff --git a/OpenTabletDriver.Daemon/DriverDaemon.cs b/OpenTabletDriver.Daemon/DriverDaemon.cs
index e3c81679..ceefcf46 100644
--- a/OpenTabletDriver.Daemon/DriverDaemon.cs
+++ b/OpenTabletDriver.Daemon/DriverDaemon.cs
@@ -37,7 +37,7 @@ namespace OpenTabletDriver.Daemon
             Log.Output += (sender, message) =>
             {
                 LogMessages.Add(message);
-                Console.WriteLine(Log.GetStringFormat(message));
+                Console.Out?.WriteLine(Log.GetStringFormat(message));
                 Message?.Invoke(sender, message);
             };
```
Does this fix it? Kinda hard to go by without line numbers.

closed 2026-10-03 by gonX