OpenTabletDriver/OpenTabletDriver #1858: Multi-monitor mapping broken on KDE Wayland

Casul51, 2022-01-03:
## Description
OTD display settings map only selected part of the active system screen.
So, if I have my mouse on screen 1 and have mapped screen 2 in OTD, stylus will move inside screen 1, in area mapped as it was OTD mappings for screen 1. 

## System Information:

| Name             | Value |
| OTD version | git - 0.5.3.1-r1017.gd7580231+d7580231 |
| Linux: | Arch linux |
| KDE Plasma Version: | 5.23.4 |
| KDE Frameworks Version: | 5.89.0 |
| Qt Version:| 5.15.2 |
| Kernel Version:| 5.15.11-230-tkg-pds (64-bit) |
| Graphics Platform: | Wayland |

jamesbt365, 2022-01-03:
Do you have clamp input outside area enabled when you right click the tablet area?

Casul51, 2022-01-03:
[image]

When I right click in tablet area (screen on the right) I have this working area (in red). Same thing happens for other screens.
So display config in OTD works as it only for active screen.

Casul51, 2022-01-03:
In Xorg everything is OK.

gonX, 2022-01-03:
As a temporary workaround you can map to the entire virtual screen (right click Display -> Set to Screen -> Virtual Display) and define the tablet-to-screen mapping in your Wayland compositor instead.

The real fix for us (OpenTabletDriver) is likely to communicate with the compositor and tell it not to map to a specific screen.

Casul51, 2022-01-03:
Temporary workaround works.
Hope real fix would be easy to implement.

jamesbt365, 2025-01-24:
We are going to track this in #3459 instead.

closed 2025-01-24 by jamesbt365

claymorwan, 2025-02-20:
Say I'm a bit confused, has it been fixed ? I still have that problem myself and idk how to fix it