OpenTabletDriver/OpenTabletDriver #1862: Switching Presets (Profile) using a new type of binding

Mrcubix, 2022-01-03:
## Description

- A new type of Binding should be implemented:
    - A new Type of Binding: Preset
    - A List of existing Presets should be available as properties

# Potential Problems:

- Property list might not be updated in real time at the moment (require testing)

X9VoiD, 2022-01-04:
Preset switching do not need a new type of binding if it's going to be the only binding dealing with presets. `IStateBinding` is enough for such purposes.

You most likely meant "Allow plugins to change current active preset" for the daemon side, and "Allow plugins to declare Preset property" for the UX side.

InfinityGhost, 2022-11-10:
This has been merged with the DI rewrite.

closed 2022-11-10 by InfinityGhost