OpenTabletDriver/OpenTabletDriver #1870: no settings applied after resume from machine suspension/hybernation

OniCado, 2022-01-05:
## Description

after waking up the machine from suspension/hybernation, OTD will not send inputs unless settings are reapplied (this does not affect v0.5.3.3)
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 21H2 10 Pro N Build 19044.1415 |
| OpenTabletDriver Version | 0.6 prerelease/master |
| Tablet | --- |

InfinityGhost, 2022-01-07:
There's no reason this should occur, nothing has changed there since v0.5.3.3. I'm unable to reproduce this.

gonX, 2022-01-18:
Please bisect for this.

Does the tablet debugger still work when coming out of suspend?

Sinamer, 2022-02-03:
Can confirm, I'm experiencing the same issue. After suspending and waking up my machine, tablet input not reacting to any actions. Clicking `Apply` in GUI makes it work again. This issue is not present in v0.5.3.3. Any version above it has the same behaviour.
Here's a clip of tablet debugger after a wake up (notice how the OS cursor is not moving):

https://user-images.githubusercontent.com/53346335/152315845-10215416-0770-4509-99a4-432d2e063675.mp4

Mrcubix, 2022-02-03:
After learning a bit more about this, i think you should try one of the following:

- Enable USB > USB selective Suspend in windows advanced power settings
- Find an option in your bios to keep USB ports powered when off or suspended

Sinamer, 2022-02-03:
Thanks for the guidance. Have tried both just now. For the former, USB selective suspend setting in power options doesn't have any effect. For the latter, there's no such USB configuration option in the UEFI for me.
I don't think it is related to the way my machine is set up, since this issue is not present in a different version of OTD.

Mrcubix, 2022-02-03:
any OTD versions have the same issue as far as what have been reported in the past and what i'm aware of

Sinamer, 2022-02-03:
OTD v0.5.3.3 does not show this behaviour and works after suspension. The original post also mentions that. Am I missing something?

Mrcubix, 2022-02-03:
#552

AkiSakurai, 2022-02-03:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/707f71ec9399056dddd8cfb6d94d0680b0b1c9f5/OpenTabletDriver.Daemon/DriverDaemon.cs#L66-L70
Is SetSettings missing here?

vedattt, 2022-02-03:
@OniCado @Sinamer Grab an action build from #2041, see if it fixes the issue.

Sinamer, 2022-02-03:
Yup, that fixes the issue! Thank you for looking into it!

PR #2041 (2022-02-06): DriverDaemon: Call SetSettings on wake from sleep
This might fix the issue with tablets not being detected after sleep. Needs to be tested.

## Changes

- Call `SetSettings` after detecting tablets when waking from sleep.
- Change sleep detection log level to `Info`, as `Debug` is particularly odd now that sleep detection is kept out of debug builds.

## Issues

- Closes #1870
- Closes #2044 

---

Edit: Seems to work: https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1870#issuecomment-1028945702
closed 2022-02-06 by InfinityGhost