OpenTabletDriver/OpenTabletDriver #1874: Support Request: [DllNotFoundException:Error] Unable to load shared library 'librt' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibrt: cannot open shared object file: No such file or directory

BubblegumZen, 2022-01-06:
# Support Request
Whenever I connect my tablet to my PC, OpenTabletDriver does detect it, but when I try to hover the pen over the tablet, it switches back to "No tablets detected", after looking at the `console` tab of the Opentabletdriver GUI, it says the following `[DllNotFoundException:Error] Unable to load shared library 'librt' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibrt: cannot open shared object file: No such file or directory`

[here is the video of the issue](https://streamable.com/i2j5v5) 

## What I've already tried
I have tried to check librt is, at best I could find a reference of `libc6` after searching for `librt` which I have already made sure is installed. In addition to that I also ran `whereis librt` and got this `librt: /usr/lib/x86_64-linux-gnu/librt.a` if this is relevant.

## Diagnostic Information
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7821016/diagnostics.txt)

hwsmm, 2022-01-06:
This is a workaround: `sudo ln -s /usr/lib/x86_64-linux-gnu/librt.so.1 /usr/lib/x86_64-linux-gnu/librt`

This issue is already fixed in master, but 0.6 is not released yet

BubblegumZen, 2022-01-06:
> This is a workaround: `sudo ln -s /usr/lib/x86_64-linux-gnu/librt.so.1 /usr/lib/x86_64-linux-gnu/librt`
> 
> This issue is already fixed in master, but 0.6 is not released yet

this worked! Thanks a lot

closed 2022-01-06 by BubblegumZen