OpenTabletDriver/OpenTabletDriver #1875: Tooltips can cause a crash when unplugging a tablet

Kuuuube, 2022-01-06:
## Description

Reproduction steps:
1. Put the cursor over and box that shows a tooltip
2. Wait for the tooltip to pop up
3. Unplug a tablet
4. Move the mouse

Error logs: 
[error1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7823717/error1.txt)
[error2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7823718/error2.txt)

Thanks to @X9VoiD for figuring out the reproduction steps.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Win10
| OpenTabletDriver Version | 0.6.0rc3

gonX, 2022-01-06:
There are reports that this did not happen on 0.5.3, and this has the same stacktrace as dotnet/wpf#5730 which is also related to a .NET 5 to .NET 6 transition.

Thanks to @vedattt for finding this

vedattt, 2022-01-06:
Unfortunately it's not entirely .NET's fault as #1750 seems to be the cause of this. Destroying tabs instead of disabling their visibility seems to make underlying objects go null hence the NRE. The fix should ideally be a way to dispose all tab contents including tooltips before getting rid of the tab.

vedattt, 2022-01-07:
Disposing tabs is not a good idea for the current implementation where the tab instances are kept as fields of the class throughout the lifespan of the class.
I've tried:
- `Suspend`ing layout upon profile change
- Modifying each tab using `Enabled = false; Visibility = false` and invalidating for a repaint before queuing (`AsyncInvoke`) another event to clear tablist and readd the new items.

These two haven't made a difference. Especially due to the second attempt, I'm suspecting that there's no easy way to play with tabs while they have a tooltip on display without things going bad in some way. Might need more testing by someone else as I'm stumped on that one.

There are two alternatives I've come up with but have not tested as they are a bit less than ideal:
- Instead of storing the tabs at all times, construct them anew upon each profile change and add those to the tablist. This would mean more tab constructions overall, but tabs don't really change all that often anyway, so debatable. But I am concerned that this may actually not work as it's still `Dispose`ing the old tab while its tooltip is shown. Honestly if this fails then tooltips are just completely broken.
- Just catch the `ArgumentNullException` and discard it when it's originating from the `ToolTipService` or whatever.
I'm not sure this even affects the user, the tooltips always seem to be able to fade away even when the exception is thrown. It's only when the tooltip has completely disappeared that the exception is thrown, and perhaps whatever is triggering it is not so critical after all.

Both of these are a bit messy, but that's all I've gotten so far.

InfinityGhost, 2022-01-07:
Being that this is a pretty niche issue and highly unlikely to occur under normal circumstances, I'm marking this low priority and moving it to the next milestone.

X9VoiD, 2022-01-12:
Since the tablet can be "disconnected" from daemon by unhandled exceptions in input pipeline, it can actually be triggered easily, especially in this early phase of 0.6.x release.

https://discordapp.com/channels/615607687467761684/628651971267788800/930711694613839874
[image]
https://discordapp.com/channels/615607687467761684/724106667964366968/930753878687907880
[image]

InfinityGhost, 2022-01-12:
Not really sure how we could handle this ourselves since this is outside of our codebase.

PR #1921 (2022-01-12): Use page list for visibility switching on macOS only
This works around both the macOS TabPage visibility issue and the tooltip crashes on WPF. Ideally, the visibility issue should be fixed in Eto.Forms instead of this mess.

Closes #1875
closed 2022-01-12 by InfinityGhost