OpenTabletDriver/OpenTabletDriver #1901: Migration Broken for some users in 0.6

Mrcubix, 2022-01-11:
## Description

It seem from what users reported that Migration is failing with the following stacktrace:

```log
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.Desktop.Profiles.AreaSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.AbsoluteModeSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.Profile.GetDefaults(TabletReference tablet)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection.<>c.<.ctor>b__2_0(TabletReference s)
   at System.Linq.Utilities.<>c__DisplayClass2_0`3.<CombineSelectors>b__0(TSource x)
   at System.Linq.Enumerable.SelectIListIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Collections.ObjectModel.ObservableCollection`1.CreateCopy(IEnumerable`1 collection, String paramName)
   at System.Collections.ObjectModel.ObservableCollection`1..ctor(IEnumerable`1 collection)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 profiles)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 tablets)
   at OpenTabletDriver.Desktop.Settings.GetDefaultProfiles()
   at OpenTabletDriver.Desktop.Settings.GetDefaults()
   at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings()
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 8+
| OpenTabletDriver Version | 0.6.0

InfinityGhost, 2022-01-12:
Need a v0.5.3 settings file that this happens on otherwise this cannot be tested.

Doksha, 2022-01-15:
[LARGE.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7875729/LARGE.txt)
This is my settings file from 0.5.3, it is supposed to be full area on a 10 inch tablet with a slightly shrunk output, but in 6.0 it is just maxed out output and input.

numericOverflow, 2022-01-21:
FYI - I had a similar issue in Win10 where OTD tray icon would disappear and app crashed whenever I hovered over the tray icon.

I followed @Mrcubix workaround steps.

After cleaning out previous version from the extracted location, then OTD started working fine again.  Just copied the settings.json file into same folder as "OpenTabletDriver.UX.Wpf.exe" and all seems good.

X9VoiD, 2022-12-31:
We're already pass 0.5.x -> 0.6.x so closing...

closed 2022-12-31 by X9VoiD