OpenTabletDriver/OpenTabletDriver #1905: Daemon crashes on startup

vaga9938, 2022-01-12:
## Description

```
[Settings:Info] Driver is enabled.
Unhandled exception. System.Net.Sockets.SocketException (98): Address already in use
    at System.Net.Sockets.Socket.UpdateStatusAfterSocketErrorAndThrowException(SocketError error, String callerName)
    at System.Net.Sockets.Socket.DoBind(EndPoint endPointSnapshot, SocketAddress socketAddress)
    at System.Net.Sockets.Socket.Bind(EndPoint localEP)
    at System.IO.Pipes.NamedPipeServerStream.SharedServer..ctor(String path, Int32 maxCount)
    at System.IO.Pipes.NamedPipeServerStream.SharedServer.Get(String path, Int32 maxCount)
    at System.IO.Pipes.NamedPipeServerStream.Create(String pipeName, PipeDirection direction, Int32 maxNumberOfServerInstances, PipeTransmissionMode transmissionMode, PipeOptions options, Int32 inBufferSize, Int32 outBufferSize, HandleInheritability inheritability)
    at OpenTabletDriver.Desktop.RPC.RpcHost`1.Run(T host)
    at OpenTabletDriver.Daemon.Program.Main(String[] args)
    at OpenTabletDriver.Daemon.Program.<Main>(String[] args)
```
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |Garuda Linux
| OpenTabletDriver Version |0.6.0

X9VoiD, 2022-01-12:
Not an OTD issue. Restart your PC, if that did not work then empty `/tmp` or delete files in it that kinda looks like `/tmp/CoreFxPipe`.

closed 2022-01-12 by X9VoiD

vaga9938, 2022-01-12:
> Not an OTD issue. Restart your PC, if that did not work then empty `/tmp` or delete files in it that kinda looks like `/tmp/CoreFxPipe`.

Rebooted, updates came and everything worked. I apologize for the inattention.

MathisP75, 2025-02-04:
This is an issue every time you switch users under Linux. For some reason, the daemon doesn't seem to end properly, despite no process being visibly running.