OpenTabletDriver/OpenTabletDriver #1927: Add support for Huion H320M

PrincessFlu, 2022-01-13:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | HUION |
| Product Name    | Inspiroy Ink H320M |
| Width           | 10.2" |
| Height          | 6.2 |
| Pressure Levels | 8192 |
| LPI             | 5080 |

## Diagnostic Information

[asd.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7861112/asd.txt)

Mrcubix, 2022-01-13:
look simple enough, lemme make a config real quick, also, which version are you using at the moment? 0.5.3.3 or 0.6

PrincessFlu, 2022-01-13:
> look simple enough, lemme make a config real quick, also, which version of OTD are you using at the moment? 0.5.3.3 or 0.6

0.6 atm

jamesbt365, 2022-01-13:
tablet > device string reader

9580 top box
109 middle box
bottom box leave empty

press dump all, send that file here

PrincessFlu, 2022-01-13:
> tablet > device string reader
> 
> 9580 top box 109 middle box bottom box leave empty
> 
> press dump all, send that file here

[qwertyuiop.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7862981/qwertyuiop.txt)

Mrcubix, 2022-01-13:
[H320M.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7862995/H320M.zip)
Create a `Configurations` folder and put that in
then in OTD, `Tablets > Detect Tablet`, it should be detected, lemme know when done so we can proceed with the rest

PrincessFlu, 2022-01-13:
> [H320M.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7862995/H320M.zip) Create a `Configurations` folder and put that in then in OTD, `Tablets > Detect Tablet`, it should be detected, lemme know when done so we can proceed with the rest

just checking so i dont do anything wrong lol, put the entire zip into the config folder, not just the json?

Mrcubix, 2022-01-13:
the json only

PrincessFlu, 2022-01-13:
> the json only

got it, it was detected

Mrcubix, 2022-01-13:
- go to `Tablets > Tablet debugger`, 
- move your pen around, data should appear in there,
- Move your pen to the bottom right and try to the get the highest value you can for position
- Do the same but for pressure
- State if the tablet has aux keys and how many it has if it does

Mrcubix, 2022-01-13:
[H320M.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7863065/H320M.zip)
try this one, same steps

PrincessFlu, 2022-01-13:
okay, everything is working now, it has 11 keys but i have a small problem. everything is SUPER glitchy, i have 0 idea why

jamesbt365, 2022-01-13:
it is glitchy because the configuration is not complete, go to tablets > tablet debugger and get the max x and y at the bottom right of your tablet

PrincessFlu, 2022-01-13:
alright, ive gone to the bottom right of my tablet and ive done the max pressure thing, what now?

PrincessFlu, 2022-01-13:
ahh lol (also by bottom right of tablet you do mean of the tablet, not the blue square that isnt the right area right lol)

jamesbt365, 2022-01-13:
Ignore the visualizer, it is wrong because the config isn't complete, i mean the physical tablet.

Mrcubix, 2022-01-13:
i doubt the 45708 is right

PrincessFlu, 2022-01-13:
[image]

PrincessFlu, 2022-01-13:
the real y is 28580 if i go to the bottom completely

Mrcubix, 2022-01-13:
the Y is correct, the X, is not, so move your pen to the right side

Mrcubix, 2022-01-13:
[H320M.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7863217/H320M.zip)
replace the old config with this one, make sure everything work as intended (except touch if it even support it)

PrincessFlu, 2022-01-13:
still a little buggy, the cursor keeps glitching but only when i use the pen

Mrcubix, 2022-01-13:
wdym by glitching, 

if you mean teleport between 2 locations, kill or uninstall your other driver

PrincessFlu, 2022-01-13:
(also, just saying this rq: the PEN has 2 buttons, the tablet has 11. idk if that was specified lol)

PrincessFlu, 2022-01-13:
also, yeah i still have my huion driver installed, uninstalling rq

PrincessFlu, 2022-01-13:
ALMOST everything works, but it seems the bottom is set too high. the bottom of the tablet stops a bit before the bottom of the screen

PrincessFlu, 2022-01-13:
i just resized full area, it works fine now

PrincessFlu, 2022-01-13:
alr, tysm, sorry for being a tad bit dense during this lol

PrincessFlu, 2022-01-13:
just kinda tested it a bit, everything works much better now and i see no issues. thank you for helping :D

PR #2362 (2022-09-03): Add support for Huion H320M
Verification: https://discord.com/channels/615607687467761684/1001425338515071077/1001438783759204462

fixes #1927