OpenTabletDriver/OpenTabletDriver #1928: New 420 Identifier conflict with common H420 firmwares

Mrcubix, 2022-01-13:
## Description

Title state everything, Plug in a H420 with a firmware that has the following string on the following index will cause a conflict:

- 6: "H420"
- 121: "H850_F210"

The regex used in the "new" 420 identifier obviously doesn't work and therefore should either be changed & tested or permanently removed

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Any
| OpenTabletDriver Version | 0.6.0.1
| Tablet                   | H420

X9VoiD, 2022-01-13:
Will re-open once an actual case of the issue pops up. 420's index 6 regex matches an empty string, H420 is not empty.

Mrcubix, 2022-01-13:
it seem in reality to be a bigger problem than anticipated

https://discord.com/channels/615607687467761684/615611007951306863/931247058747265035

- First, i sent a config with the second identifier removed to someone, they report back that both are still detected after re-plugging,
- I then sent another config, but this time, the first, original identifier i removed, the config still matches the identifier and make use of it even if the string doesn't match,
- Finally, i sent a config with both identifiers removed, the result is still the same, even after restarting OTD.

In conclusion, 2 new issues appeared since rc4 :

- Configuration Overrides is broken
- Identifier Matching could be broken

Mrcubix, 2022-01-13:
Here is the second example of what i did to attempt to fix the issue:

https://discord.com/channels/615607687467761684/931275140036198430/931276161370177586

- Sent him the following config to put in `Configurations`
[Downloads.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7865627/Downloads.zip)

- Once placed inside, he re-plugged and it work

I'll go Troubleshoot another 3 cases to see if the result is consistant but so far it isn't.

tytan652, 2022-01-13:
[image]
Maybe it's not my regex that is wrong, but the 420 (as a regex) that match with H420

jamesbt365, 2022-03-09:
Is this fixed in #1975 or #2045?

closed 2022-03-09 by X9VoiD

tytan652, 2022-03-09:
#1935 and #1975 shall have fixed this identifier issue.