OpenTabletDriver/OpenTabletDriver #1944: Support Request:

LiftedStarfish, 2022-01-14:
# Support Request

I cannot for the life of me figure out why, I just know that it was working fine before 6.0.0
Artist mode causes the pen buttons for my Huion H950P to become non-functional on Manjaro KDE. Also, there does not seem to be a 'pen pass-through' option for me anywhere. This has lead my to a situation where I either get pen pressure but no MMB or RMB, or I get MMB and RMB, but a no pen pressure/binary pen pressure. Because most artistic programs use a great deal of both MMB and RMB for navgation, you can see why this might be an issue, and not having pen pressure basically removes the most important part of any drawing pad. Note that, prior to (and immediately after) having these issues, there were no plugins installed.

## What I've already tried

There were no wiki pages, but I was previously using the '-git' version from the AUR, but I tried switching to the non '-git' version, and that did not fix my problem.
As previously stated, I am unable to find an option for 'pen passthrough' on any of the bindings for the pen, and am thus unable to just use 'Absolute Mode' with the pen bound to passthrough. Linux Artist Mode input type only offers "Pen Button 1/2/3"
Restarting my machine multiple times 
Unplugging the device, restarting OTD, then plugging it back in.
Unplugging the device, restarting Machine, and plugging device back in.
Uninstalling all plugins (all having been installed since discovering the issue) has not solved this issue, even after having switched to non-git.
Switching to another USB Port.

## Diagnostic Information

```
{
  "App Version": "OpenTabletDriver v0.6.0.0+496f38e",
  "Build Date": "1/14/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.15.12.1",
    "VersionString": "Unix 5.15.12.1"
  },
  "Environment Variables": {
    "USER": "[REDACTED]",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "/home/[REDACTED]/.local/bin:/home/[REDACTED]/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/home/[REDACTED]/.dotnet/tools:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/var/lib/snapd/snap/bin"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:0B05:1822.001C/hidraw/hidraw14",
      "Manufacturer": "Sunrex/JME",
      "ProductName": "Ghost Key Elimiantion Keyboard",
      "SerialNumber": "",
      "FriendlyName": "Ghost Key Elimiantion Keyboard",
      "VendorID": 2821,
      "ProductID": 6178,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:1B1C:1B89.0008/hidraw/hidraw7",
      "Manufacturer": "Corsair",
      "ProductName": "CORSAIR K95 RGB PLATINUM XT Mechanical Gaming Keyboard",
      "SerialNumber": "[REDACTED]",
      "FriendlyName": "CORSAIR K95 RGB PLATINUM XT Mechanical Gaming Keyboard",
      "VendorID": 6940,
      "ProductID": 7049,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.2/0003:1532:0067.0007/hidraw/hidraw6",
      "Manufacturer": "Razer",
      "ProductName": "Razer Naga Trinity",
      "SerialNumber": "[REDACTED]",
      "FriendlyName": "Razer Naga Trinity",
[... 1623 lines cut ...]
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-01-14T07:21:02.2162704-05:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-01-14T07:21:02.2162816-05:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

gonX, 2022-01-14:
In Linux Artist Mode, you should unbind the pen tip and bind the barrel buttons. It should look like this:

[image]

Since #1736, the threshold slider no longer does anything in artist mode, ie. we do not rewrite pressure. This is similar to the stock Linux drivers, but will be a bit more obvious from the UI side of things in future versions - either by removing the bindings, or by reimplementing the pressure rewrite.

We did look at having a pen tip binding, but decided against it since it was effectively not doing anything.

Using anything other than pen buttons while in Artist Mode is unsupported, as the mouse buttons will be sent on a separate endpoint which might send invalid position coordinates.

LiftedStarfish, 2022-01-14:
This has solved my issue. Thank you! You might wanna add this to the wiki (IDK what the policy is for users like myself doing so), because I'm sure others will have the same issue in the future, and it's much easier/more intuitive to look at a help wiki than it is to sift through issue articles. Thank you, again.

closed 2022-01-14 by LiftedStarfish