OpenTabletDriver/OpenTabletDriver #1947: Relative mode no movement with very low reset times.

jamesbt365, 2022-01-14:
## Description

When reset time is set to 00:00:00 the cursor doesn't move, this should be either warned about or something should be put into place so that super low times cannot be allowed, see  [here](https://discord.com/channels/615607687467761684/615611007951306863/931609986394628116).
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | N/A
| OpenTabletDriver Version | N/A
| Tablet                   | N/A

gonX, 2025-01-24:
Re-targeted to 0.6.5.1 but should be considered for all branches.

gonX, 2025-01-25:
I see 2 ways about "fixing this":

- Tracking resets vs outputs - if e.g. 10 resets happen without doing a single output, we can relatively safely warn the user. This is a tablet agnostic solution.
- Limiting the reset time to a minimum of 20ms (50hz), though this would be slightly breaking for users using less than this. This assumes the slowest reporting tablet is 50Hz.

gonX, 2026-09-25:
Closing as this is fixed in 0.6.x with #3750 and the PR has a forward-port label should anyone wish to tackle this directly in `avalonia`.

PR #3750 (2025-01-29): [06x] RelativeOutputMode warn on reset spam
Fixes #1947 for 0.6.x. Will warn the user if 10 consecutive resets have been hit. Unsure if this should error instead.

Alternative fixes could have been calculating a ratio, maybe >10%? It would help detect resets times that are marginally too low, especially in the case of 200hz tablets that alternate between 4 and 6ms reports (as seen in 200Hz USB1.1 Wacoms).

~~Enabling data validation in the UI seems like a larger undertaking: https://gist.github.com/cwensley/8348615c67618dc46102~~ We can actually just check in `ApplySettings()` if needed. Propa [... cut]
closed 2026-09-25 by gonX