OpenTabletDriver/OpenTabletDriver #1953: Tablet not connecting after cleaning pc

deadcupcake2112, 2022-01-15:
## Description

I have tried absolutely everything to make my tablet connect, but it was working fine last night and now it won't connect to the driver. I have cleaned up my pc from any other drivers, I have reinstalled the driver, I have unplugged the tablet and plugged it back in and still nothing. It now goes to the top left of the screen no matter what I do. I have previously had issues with this as well, I don't know if I have a faulty tablet but this is aggrivating. I have been dealing without the driver for a week or two and then when I finally get it set up, the next day it doesn't work. Please help.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows
| OpenTabletDriver Version | 0.5.3.3
| Tablet                   | Gaomon S620

jamesbt365, 2022-01-15:
try this: https://github.com/X9VoiD/WinUSBCleanup/releases/tag/v1.0

if still no you have deleted the configurations folder which is required for 533

closed 2022-01-15 by jamesbt365

deadcupcake2112, 2022-01-15:
I am confused. what do you mean "file > reset to default" how do I do that? sorry I am realyl bad with technology

Mrcubix, 2022-01-15:
[image]

deadcupcake2112, 2022-01-15:
@Mrcubix it still says no tablet detected

jamesbt365, 2022-01-15:
help > export diagnostics, save the file as a txt and send it here

deadcupcake2112, 2022-01-15:
[diagnosticinformation.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7875502/diagnosticinformation.txt)

Mrcubix, 2022-01-15:
first of all, you tablet isn't even plugged in, second of all, your configurations folder doesn't exist (probably cause of OneDrive, get rid of it)

deadcupcake2112, 2022-01-15:
OneDrive? How do I get rid of that. and i dont know any good drivers. i tried gaomon driver and it didnt work for osu. it would go to the top left of the screen

deadcupcake2112, 2022-01-16:
[diagnosticinformation.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7876136/diagnosticinformation.txt)

jamesbt365, 2022-01-16:
The config folder doesn't exist, that is all i can say, you need to remove it from one drive if its doing something wack, thats all i can say. Alternatively you can use the latest version of otd which doesn't need a configurations folder due to it being in the exe itself.